indilib/indi #1780: Instructions for building the client library for Android and iOS

btschumy, 2022-12-05:
The INDI docs say that INDI clients can be written for Android and iOS.  However, there is no documentation on how to build the client library for those platforms.  Can we please get instructions in the README.md (or somewhere else) for how to do this?

Thanks,
Bill

knro, 2022-12-06:
There *was* support to build the client library on Android. However after the recent changes by @pawel-soja , I didn't check. iOS was never supported, and not sure what it entails to export the C++ library so that it can be used in iOS environment if it is at all possible like with Android.

btschumy, 2022-12-06:
On the main INDI documentation page at: https://docs.indilib.org/

It says:

While the INDI wire protocol is platform-independent, the official INDI Library is designed to operate specifically on POSIX platforms. Developers can port the library and device drivers to different platforms as desired.

Supported operating systems include:

Linux: Full Support
MacOS: Full Support, except for few Linux-only drivers.
Windows: Partial driver support via Cygwin. Client support.
BSD: Full Support, except for few Linux-only drivers.
iOS: Client support only.
Android: Client support only.
If you don’t support iOS clients, then this should be removed.  My guess is that you can compile the client library for iOS.  I was able to do it for INDIGO although it took a bit of finagling.  It required creating a Xcode project for it and adding the appropriate files.  I might try that if there is not official support.

If there is not Android support either, I will try building it in Android Studio.

Bill

> There was support to build the client library on Android. However after the recent changes by @pawel-soja <https://github.com/pawel-soja> , I didn't check. iOS was never supported, and not sure what [...]

pawel-soja, 2022-12-06:
That's right, I reorganized the project structure and edited the `CMakeLists.txt` files.
For Android, the `indiclientqt` library should build (if it was building before the changes).

I'd like to help, but I'm not currently working on Android and iOS. If there is a lot of interest in these systems, when I have some time, I will introduce them to CI and prepare compilation scripts.

btschumy, 2022-12-06:
Pawel,

Thanks for replying.

1. I am not using Qt.  Does this build target also build the non-Qt version?
2. I assume I need to install Qt in order to build the indiclientqt lib?
3. Do you know if I can build this on macOS or do I need a Linux system?

Bill

> That's right, I reorganized the project structure and edited the CMakeLists.txt files.
> For Android, the indiclientqt library should build (if it was building before the changes).
>
> I'd like to help, [...]

knro, 2023-02-27:
There is no proper Android/iOS client support unless the C++ client can be used as-is. Another developer is working on Rust bindings but it's not ready. A javascript client library would be ideal.

closed 2024-04-08 by github-actions[bot]