indilib/indi #1796: Unrestricted access to /dev/ttyUSB* vulnerability

dimich-dmb, 2022-12-26:
`drivers/auxiliary/99-indi_auxiliary.rules` sets mode `0666` for many popular USB-to-serial adapters. It allows any user to access any `/dev/ttyUSB*` unrestrictedly in most cases, regardless of `uucp` group membership and device purpose. This is security vulnerability.

Possible solution is to change `MODE="0666"` to `TAG+="uaccess"`. Probably also need to change `99` prefix in `99-indi_auxiliary.rules` to move it before `73-seat-late.rules`.

knro, 2023-01-15:
Would this work *across* distro? Many users are already suffering from being unable to access their USB devices due to all the restrictions and we don't want to make the users suffer unnecessarily.

dimich-dmb, 2023-01-16:
> Would this work _across_ distro?

I don't know. I guess it should work on any distro with standard systemd configuration but can't test it. I have libindi installed as a dependency for another package.

Anyway, letting to know about security issue is better than staying silent.

> Many users are already suffering from being unable to access their USB devices due to all the restrictions and we don't want to make the users suffer unnecessarily.

On another hand, allowing all users to access USB devices is a rude solution, i think. Let users manage access to devices on their systems.

closed 2024-04-06 by github-actions[bot]