indilib/indi #1798: Server disconnection event no longer working on 1.9.9

Amanda-M-UK, 2022-12-28:
I have recently updated my software to link against 1.9.9 and noticed that the server disconnection event no longer works. It worked correctly on 1.9.6. 

The problem exists on both Linux and Windows using INDI::BaseClient.

### Steps to reproduce the problem

**Step 1**

Link against INDI 1.9.6, connect to an INDI server and then disconnect. After doing so,  _virtual void serverConnected()_  and _virtual void serverDisconnected(int)_ are called. This is, of course, the correct behaviour. 

**Step 2**

Link against INDI 1.9.9, connect to an INDI server and then disconnect. After doing so,  _virtual void serverConnected()_  is called but not  _virtual void serverDisconnected(int)_.  This looks like a bug or perhaps a design change that I am not aware of.

It's vital to the design of my software that _virtual void serverDisconnected()_  is called after the INDI server disconnects.  Can anybody help with this problem?

Amanda

knro, 2022-12-28:
@pawel-soja possible baseclient regression?

pawel-soja, 2022-12-28:
I'll check it. Although from what I remember serverDisconnect will be triggered when the server disconnects. It doesn't fire when the client closes the connection. Similar behavior as in the case of BaseClientQt using the Qt library. I may be wrong, I'll check it out.

pawel-soja, 2022-12-28:
Yes it is a bug.

The `BaseClientQt::serverDisconnected` function is called when `QTcpSocket::errorOccurred` occurs and this is called when the server disconnects, not the client.

So client disconnection should not call `BaseClientQt::serverDisconnected`.
But... now I noticed that the function is called because this is how the implementation of the `BaseClientQt::disconnectServer` function looks like:
```cpp
bool BaseClientQt::disconnectServer(int exit_code)
{
    /* ... */
    serverDisconnected(exit_code);
    /* ... */
}
```

This call is missing in the `BaseClient::disconnectServer` function.

Before refactoring there was spaghetti in `BaseClient`, so I followed the behavior of `BaseClientQt` and I missed this difference.

I will prepare a fix soon.

Amanda-M-UK, 2022-12-28:
This is great news, thank you. I'll await the fix.

pawel-soja, 2022-12-28:
Fixed: https://github.com/indilib/indi/pull/1799

Please check and confirm if everything is ok.
Thank you!

Amanda-M-UK, 2022-12-28:
Hi,

I grabbed the master branch again but it didn't contain the fix. However,  I manually applied the fix and can confirm that it's now working.  The fix I applied is below. 

Which branch should I have checked out?

```
bool BaseClient::disconnectServer(int exit_code)
{
    D_PTR(BaseClient);

    if (d->sConnected.exchange(false) == false)
    {
        IDLog("INDI::BaseClient::disconnectServer: Already disconnected.\n");
        return false;
    }

    d->exitCode = exit_code;
    d->clientSocket.disconnectFromHost();

    serverDisconnected(exit_code); // Added this line

    return d->clientSocket.waitForDisconnected();
}
```

pawel-soja, 2022-12-28:
Changes have not yet been committed to the master branch. The fix is available on branch:
https://github.com/indilib/indi/tree/fix/baseclient-disconnected
According to the pull request: https://github.com/indilib/indi/pull/1799

I'm glad this type of fix works for you.

Amanda-M-UK, 2022-12-28:
It does, thanks for your efforts. 

I have found another problem when building INDI on MSYS2 / Mingw. I'll make a new bug report with more details.

pawel-soja, 2022-12-28:
Please check the master branch.
Changes have just been accepted, also related to Cygwin.

pawel-soja, 2022-12-28:
I'm closing the issue.
Please let me know if you have any further problems with this topic.

Thank you very much!

closed 2022-12-28 by pawel-soja