indilib/indi #1800: Building INDI 1.9.9 on MSYS2 / Mingw

Amanda-M-UK, 2022-12-28:
When building INDI 1.9.9 on Windows using the MSYS2 environment the build fails at line 49 of indiutility.cpp.

The  code block in question is as follows:

```
#ifndef _WINDOWS
int mkpath(std::string s, mode_t mode)
{
    size_t pre = 0, pos;
    std::string dir;
    int mdret = 0;
    struct stat st;

    if (s[s.size() - 1] != '/')
        s += '/';

    while ((pos = s.find_first_of('/', pre)) != std::string::npos)
    {
        dir = s.substr(0, pos++);
        pre = pos;
        if (dir.size() == 0)
            continue;

        if (stat(dir.c_str(), &st))
        {
            if (errno != ENOENT || ((mdret = mkdir(dir.c_str(), mode)) && errno != EEXIST)) // Build fails here
            {
                return mdret;
            }
        }
        else
        {
            if (!S_ISDIR(st.st_mode))
            {
                return -1;
            }
        }
    }
    return mdret;
}
#endif
```

The problem is with mkdir(dir.c_str(), mode)). It appears that under MSYS2 / Mingw or MSYS2 / Clang, mkdir() only accepts one parameter. I fixed this by removing _mode_ from mkdir(). Once I did that, INDI built correctly and appears to work. 

If you need any further information then please let me know. I'm happy to test whatever code you change. 

Amanda

pawel-soja, 2022-12-28:
I need to think about adding Cygwin to CI. I'm a bit of a guess without owning this platform.

The function `mkpath/mkdir` was probably not used when compiling for the Windows platform.

Please try: https://github.com/indilib/indi/pull/1801
**Branch:** fix/win32-disable-mkdir

Amanda-M-UK, 2022-12-28:
No rush, as changing the parameter for mkdir() allowed me to build INDI.

If you want me to test anything on my MSYS2/Mingw environment then just ask. Generally, I code on Linux and then build for other platforms. However, I have quite a bit of experience with Mingw so happy to help where I can.

Amanda

pawel-soja, 2022-12-29:
I initially ran the build in CI on MSYS2 Mingw and reproduced the problem.
I also see that the problem is the compilation of the `sharedblob.c` file, right?

pawel-soja, 2022-12-30:
Fixed: https://github.com/indilib/indi/pull/1802

Please try if everything is ok.

Amanda-M-UK, 2022-12-30:
I can confirm that this problem has been fixed. I wasn't aware of a problem in sharedblob.c. 

Thank you very much.

Amanda

closed 2022-12-30 by pawel-soja