indilib/indi #1808: serverDisconencted(int exit_code) runs for failed INDI server connection

Amanda-M-UK, 2023-01-08:
I recently submitted a bug report regarding the serverDisconencted(int) event not running after disconnecting from an INDI server.  This problem was fixed, thank you. However, another problem has come to light regarding this disconnect event. 

It now appears that the serverDisconencted(int) event runs (with an exit code of -1) even if a connection to an INDI server initially fails. This did not happen in version 1.9.6.

### Steps to reproduce

If the following code is run with the host parameter set to an IP which does not have an INDI server running then the returned bool is false (as it should be) but serverDisconnected(int) is called. 

`setServer(host, port);
bool state = connectServer();`

As mentioned above, serverDisconencted(int) was not called in version 1.9.6 for failed connections but is for version 1.9.9 and 2.0.0. Unfortunately, I have designed my application to do some housekeeping after an INDI server is shutdown or otherwise goes off line for some reason. If serverDisconnected(int) is called on an initially failed connection this causes early deallocation of some of my resources which is causing a problem. 

Is it possible to stop serverDisconnected(int) from being called if a user accidentally enters an incorrect IP? If this is not possible, could serverDisconencted(int) have a different exit_code?

Amanda

knro, 2023-01-26:
@pawel-soja Any ideas about this issue?

pawel-soja, 2023-01-27:
The proposed fix in pull request does not call the function if there is no connection.

In fact, the behavior has changed.
I didn't notice that this case is unsupported (`BaseClientQt` / `sConnected == false`).
https://github.com/indilib/indi/blob/002040d7edaafe1a62cd2ef98a05714d022fa53d/libs/indiclientqt/baseclientqt.cpp#L96-L112

My goal was to introduce an asynchronous connection because currently connecting is blocking the thread. The serverDisconnected function was supposed to return an error status.
Internally, the library is prepared for such a transition, but it poses a huge risk of backwards incompatibility.

In the case of `QTcpSocket`/`INDI::TcpSocket` it is possible to read the error value (`SocketError`).
```cpp
    clientSocket.onErrorOccurred([this] (TcpSocket::SocketError)
    {
        if (sConnected == false)
            return;

        this->parent->serverDisconnected(-1);
        /* ... */
    });
```
But in the `INDI::BaseMediator` class the function argument is problematic, some enum would be nice.
```cpp
        /** @brief Emmited when the server gets disconnected.
         *  @param exit_code 0 if client was requested to disconnect from server. -1 if connection to server is terminated due to remote server disconnection.
         */
        virtual void serverDisconnected(int exit_code);
```

closed 2023-02-04 by pawel-soja