indilib/indi #1815: The 2 libindi MathPlugins from the Fedora37 Bleeding Repository appear to have been compiled against the wrong indi library

ebessant, 2023-01-21:
Since 2023-01-08, my "dnf upgrade" jobs have been failing for "indi-*" and "kstars" from the "copr:copr.fedorainfracloud.org:xsnrg:libindi-bleeding" and "copr:copr.fedorainfracloud.org:xsnrg:indi-3rdparty-bleeding" repositories have been failing.  These failures h
ave been down to the 2 Math Plugins "libindi_Nearest_MathPlugin.so" "libindi_SVD_MathPlugin.so" needing the file "libindi-libs-1.9.9-1.fc37.x86_64" to be installed.  Alas, this file (from the Fedora updates repository) conflicts with the "indi-libs-2.0.0" file fro
m the "libindi-bleeding" repository.

If I run the command "dnf -x libindi-libs upgrade" then "kstars" and all of the "indi-*" files are excluded from the upgrade process
.

Attached is text file containg the output from the "dnf upgrade" command showing the actual error messages.

I know that I can download all of the source code from git and compile everything myself, but the "dnf upgrade" process is far more
convenient and lets me know precisely the versioning of everything that is installed.

Many apologies for the 11-day delay in reporting this, but my anxiety issues at the moment make if very difficult to open any dialog
ues in any medium, and I need time to build up my courage to start the ball rolling.  Thank goodness for my telescope which helps me
 keep some form of sanity.

Thanks for your time and understanding,
[indi-Math-Plugins-Upgrade-Failure.txt](https://github.com/indilib/indi/files/10472927/indi-Math-Plugins-Upgrade-Failure.txt)

Ed Bessant.

**Desktop (please complete the following information):**
 - OS:  Fedora 37
 -         Gnome 43
 -         Kernel - 6.1.6-200.fc37.x86_64

**Log Files**
Make sure to enable logging and include log files
https://indilib.org/support/logs-submission.html
[indi-Math-Plugins-Upgrade-Failure.txt]
(https://github.com/indilib/indi/files/10472933/indi-Math-Plugins-Upgrade-Failure.txt)

knro, 2023-01-22:
@xsnrg Can you please check this issue?

xsnrg, 2023-01-22:
This is an odd problem. At some point along the way, the main Fedora repo indi library was pulled in.  What I had to do on my own system was remove indi, and then re-install again from the copr repo. It should pull indi-2.0.0-git now and install everything from copr. Perhaps the math libraries had a bad compile with a weak link to the indi library and that was fixed?

At any rate, try this, and if everything is good for you as well, please close this. I hope it was transient.

ebessant, 2023-01-22:
Firstly, thanks for getting to this so quickly.  Such responses are always refreshing.

I followed your instructions, and removed all of the modules listed in the failed "dnf upgrade" process.  I then immediately followed that with a "dnf install" process with all of the above-mentioned modules PLUS "indi-eqmod" and "stellarium" which were also removed as"dependent packages".

Alas, this clean installation process failed with the same errors for the 2 Math Plugins.  I then looked at the output for the clean installation and noticed that the copr module "indi-eqmod" was still on version "1.9.9.git-20221201101130.fc37" rather than the "2.0.0.git-2023*fc37" version for all of the other "copr:indi*" modules.  Removing both "indi-eqmod" and "stellarium" from the clean installation list sorted out the issue for me.  The errors from "dnf" regarding the Math Plugins are rather misleading, since it now looks like the problem is down to the "indi-eqmod" not being compiled  with the new version of "indi"

Since I use an iOptron CEM60 mount, I am fairly sure that I can live without the "indi-eqmod" SkyWatcher module.  With regard to "Stellarium", I would be happier with it still on my computer, but since it is part of the Fedora repositories, I will wait until the bleeding version is made live across to Fedora.

Attached is a text file showing the outputs from the 6 steps taken with "dnf" to get down to the real issue.

I am not sure whether to close this at the moment, since, even though my problem is sorted, there is still the "indi-eqmod" module problem to be sorted.

Thanks again, and if you are ever looking for someone to do some Fedora testing for you, I am always here.

Ed Bessant.

[remove-indi_then_reinstall-indi.txt](https://github.com/indilib/indi/files/10474009/remove-indi_then_reinstall-indi.txt)

xsnrg, 2023-01-22:
It looks like the CMakeFile was updated to include a new dependency that is not found automatically, and not listed in the spec file. I will see if I can get a PR up so 2.0.0 eqmod gets a clean build

xsnrg, 2023-01-22:
@iliaplatone I missed it, but eqmod has been failing in the COPR builds since 12/7/22, with the AHPGT things missing. I will try to see if it is as simple as adding a package, but it looks like the whole thing is in an `if(WITH_AHP_GT)`  that is failing from the CMakeFile.

```CMake Error at /builddir/build/BUILD/indi-3rdparty-master/cmake_modules/FindAHPGT.cmake:52 (message):
  AHPGT not found.  Please install libahp_gt-dev
Call Stack (most recent call first):
  CMakeLists.txt:104 (find_package)```

xsnrg, 2023-01-22:
OK, it is a whole new library that was added that needs to be included. Should have a fix shortly

xsnrg, 2023-01-22:
https://github.com/indilib/indi-3rdparty/pull/708

xsnrg, 2023-01-23:
@ebessant is your problem fixed now?

ebessant, 2023-01-23:
@xsnrg :  Firstly, thanks for the work yesterday.  Yes, all has been sorted now.  I reinstalled the eqmod module last night after you had found the missing library, both the module and new library got installed happily.  This morning's big update of everything indi, indi-lib and kstars went through successfully as well.

As an interim until Stellarium is built with this newest version of libindi-libs, I have set up a virtual box devoted solely to Stellarium.

Thanks again.

E.

closed 2023-01-23 by ebessant

ebessant, 2023-02-03:
This is just an addendum to the issue.  After my fortnightly check on the GA state for for indi-2.0.0, I noticed today that it has been released.  As a result, I tried to re-install Stellarium on one of my Fedora 37 machines, and SUCCESS!!!  I once again have all my astro software on the same machine.

Thanks to all of you for your exemplary work on this wonderful package.

E.