indilib/indi #1822: FAST BLOB possibly broke BLOB Snoop

knro, 2023-01-26:
Tried to used astrometry driver with Ekos in remote astrometry mode and noticed that snooped camera BLOB was always empty, is the BLOB snoop mechanism no longer working @pludov ?

pludov, 2023-01-26:
i dont remember having tested it ... I will look into it.

knro, 2023-01-29:
@pludov Thank you! Any update on this issue? INDI v2.0.0 is due on Tuesday and this is a show-stopper bug.

pludov, 2023-01-29:
Reading attached blob is simply not implemented on the driver side (see the fixme in indidriver.c: 433).

This needs implementation in indidrivermain.c, indidriver.c, so probably not something to implement in a rush (not for Tuesday for sure !)

So I pushed a "workaround" that simply rollback to base64 for blobs sent to drivers via snoop (only on the server -> snooping driver path)

closed 2023-01-30 by knro