indilib/indi #1829: Astroelectronic FS2 - Mount moves to wrong position

funastro, 2023-01-31:
**Describe the bug**
Mount can be correctly selected in KStars. The position is accepted correctly. However, after selecting a target, the mount moves to the wrong position and can only be stopped using the hand box

**To Reproduce**
Exact steps to reproduce the behavior.
1. Start Stars with EKOS
2. Start in EKOS Profile
3. Mount Position accepted
4. Select visible object in map
5. Click slew
6. Mount is running in a false direction and don't stop

**Expected behavior**
Mount running to the correct position

**Screenshots**
No screenshots

**Desktop (please complete the following information):**
 - OS: MAC OS Ventura 13.1

**Log Files**
No Log Files today, bad weather.. My idea: Will the LX200 - Basic Syntax be changed? Especially with the commands :Sr bzw. :Sd

m-butti, 2023-07-19:
I would like to help fixing this issue.

Currently we are making a custom build of Stellarium with "fixed" LX200 implementation to make it work with FS2 when ever there's a new release.
We now updated to the latest FS2 firmware 1.26 with ACK command support to make it usable via INDI so we don't have to build a custom Stellarium anymore. Unfortunately we encountered this issue with INDI too.

A probable cause might be, that FS2 needs a space character after the :Sr and :Sd commands.
I would like to verify this. Is there a way to log all commands INDI sends to the COM Port?

knro, 2023-07-21:
Absolutely, you can log all communication via Options --> Debug and you'll see it in the log file. It would be great if this is fixed before next release of INDI next week.

m-butti, 2023-07-22:
Thanks @knro for the swift reply

Today I could dig into it a bit more.
This is what I found out:

- Some space needed
The :Sr and :Sd command need a space character to work properly.

The commands for the slew sent by INDI as I have recorded them:
:Sr07:15:10#:Sd+09*57:08#:MS#
It would work with FS2 if it looked like this:
:Sr 07:15:10#:Sd +09ß57:08#:MS#

- Don't tell a stopped FS2 to stop
From the second time i wanted to initiate a GoTo INDI (Or kstars?) added a stop command (:Q#) in front:
:Q#:Sr06:30:39#:Sd+28*27:12#:MS#
Unfortunately FS2 when not in motion seems to buffer this command and executes it the next time it is in motion.
This leads to an immediate abort of the movement after the :MS# triggers and the telescope does not get to move at all.
For FS2 we should never submit a :Q# instruction except when we know that it's actually moving right now or it is manually initiated by clicking a stop button in the GUI or so.

- A star as the degree symbol?
I have noticed, that INDI uses a star character (*) for the degree symbol in the :Sd (Set Declination) command.
That's how it is written in the documentation everywhere, but according to the following comment in the Stellarium source code, Maede actually uses the ANSI 223 character (ß) in their devices:
https://github.com/Stellarium/stellarium/blob/b837ea35bbaf4425a5aa8c4ddf333f456d8ba898/plugins/TelescopeControl/src/Lx200/Lx200Command.cpp#L176
I can't verify, if that's really true for Maede devices, but I see that FS2 actually answers :GD# (Get Declination) with this symbol:
+09ß15:27#
It might not be relevant what character is used after all. Don't know. :)

- Just a symlink?
When doing an "ls -l /bin/indi_lx200fs2" I noticed that it just points to /bin/indi_lx200generic.
Does that mean, that there isn't actually any FS2 specific logic in place?

- Unsupported Commands
While testing I stumbled upon the following commands, INDI has sent:
:GM# // Site name
:GT# // Track frequency
:Gt# // Site latitude
These are unsupported by FS2. Probably not a big deal and shouldn't hurt.
Supported commands are listed here: http://www.astro-electronic.de/fs2.htm

Sooo, that's quite a bit. Hope it's understandable.
Unfortunately I don't feel confident enough in C++ that I could create a complete pull request to make FS2 work (again) for the next release.
If you or someone else could make it happen, we would be very happy to be able to use INDI in our observatory.
I'm always available to test anything with FS2.

Regards Michael

commit 745b13b3 (2023-07-23): Fix FS2 GOTO issue by adding space. Fixes #1829
closed 2023-07-23 by knro