indilib/indi #1830: PMC8_indi not working for south hemisphere

nachoruiz29, 2023-02-01:
Basically the error that I saw is that the driver does not change the sign of the declination (conceptually the (+) sign is used for the northern hemisphere and the (-) sign for the southern hemisphere), in the Indi Drive of the PMC Eight it uses the (+) sign for the declination in the southern hemisphere, that generates a data reading error with the declinations with sign (-) in the southern hemisphere. That's why you can't put a signed latitude (-) (south latitude). If you put a north (+) latitude in your ASIAIR it will work correctly, but it clearly does not work to do that in the southern hemisphere.

jfwells, 2023-02-06:
Odd, I have an iexos 100 and was suffering with incorrect gotos and tracking, until I checked the geography and it had me in the southern hemisphere.  As a workaround, can you set the PMC8 mount location elsewhere -- i.e. in Windows or in ExploreStars and then set INDI not to update it?

knro, 2023-02-07:
I didn't check the code, do you have a solution for this?

nachoruiz29, 2023-02-07:
Karl, one of the main contributors for this driver is taking care of it.
His is about to send the branch for compiling and testing.
stay in the loop, a fix will come.

knro, 2023-02-27:
It's fixed in !1856

closed 2023-02-27 by knro