indilib/indi #1831: kstars broken with libindi 2 update

mgutierrezp, 2023-02-01:
kstars still searches the old version of libindi:

kstars: error while loading shared libraries: libindiclient.so.1: cannot open shared object file: No such file or directory

# ls -la /usr/lib/x86_64-linux-gnu/libindiclient*
lrwxrwxrwx 1 root root     22 Jan 31 09:46 /usr/lib/x86_64-linux-gnu/libindiclient.so.2 -> libindiclient.so.2.0.0
-rw-r--r-- 1 root root 604432 Jan 31 09:46 /usr/lib/x86_64-linux-gnu/libindiclient.so.2.0.0

knro, 2023-02-02:
This is not an INDI issue. It's probably a build issue with KStars. Try to clear KStars build directly and rebuild again.

closed 2023-02-02 by knro

mgutierrezp, 2023-02-02:
I did not build kstars from sources and there are no updates on the ppa repo. I will open a ticket accordingly. Thanks Jasem!

mgutierrezp, 2023-02-03:
> This is not an INDI issue. It's probably a build issue with KStars. Try to clear KStars build directly and rebuild again.

Jasem, sorry, where can I open a ticket for kstars-bleeding? I see a newer version in ppa repository (3.6.3) that may solve the issue, but I have a mint distro based on ubuntu 20, and there is no deb package for such version