indilib/indi #1833: Breaking changes in 2.0.0 not mentioned in release notes

paolostivanin, 2023-02-02:
Hello,
it seems that 2.0.0 is breaking compatibility with some applications. The ones I know of are Stellarium and PHD2 (the latter when compiled using system INDI).
Is there a guide somewhere on what changed and how developers can migrate to this new format that has been introduced?

Thanks

paolostivanin, 2023-02-02:
PHD2 has an option (that's not very good for packagers) to build the included libindi (a very old version, 1.7.5).
Stellarium, AFAICS, has no option to use anything else than system INDI, so it now fails with:
```
[  217s] /home/abuild/rpmbuild/BUILD/stellarium-1.2/build/plugins/TelescopeControl/src/INDI/TelescopeControl_INDI_autogen/EWIEGA46WW/../../../../../../../plugins/TelescopeControl/src/INDI/INDIConnection.hpp:72:14: error: 'void INDIConnection::newDevice(INDI::BaseDevice*)' marked 'override', but does not override
[  217s]    72 |         void newDevice(INDI::BaseDevice *dp) override;
[  217s]       |              ^~~~~~~~~
[  217s] /home/abuild/rpmbuild/BUILD/stellarium-1.2/build/plugins/TelescopeControl/src/INDI/TelescopeControl_INDI_autogen/EWIEGA46WW/../../../../../../../plugins/TelescopeControl/src/INDI/INDIConnection.hpp:73:14: error: 'void INDIConnection::removeDevice(INDI::BaseDevice*)' marked 'override', but does not override
[  217s]    73 |         void removeDevice(INDI::BaseDevice *dp) override;
[  217s]       |              ^~~~~~~~~~~~
[  217s] /home/abuild/rpmbuild/BUILD/stellarium-1.2/build/plugins/TelescopeControl/src/INDI/TelescopeControl_INDI_autogen/EWIEGA46WW/../../../../../../../plugins/TelescopeControl/src/INDI/INDIConnection.hpp:74:14: error: 'void INDIConnection::newProperty(INDI::Property*)' marked 'override', but does not override
[  217s]    74 |         void newProperty(INDI::Property *property) override;
[  217s]       |              ^~~~~~~~~~~
```

knro, 2023-02-02:
Ok let me see if I can fix that for them. Just forked Stellarium but it's taking forever to checkout.

pawel-soja, 2023-02-02:
Since version 2.0.0 there is a new interface that is not based on raw pointers. Instead, the INDI::Property/INDI::BaseDevice classes are used to properly manage their pointer copies to ensure memory access security.

tstibor, 2023-02-02:
Note there is already a discussion on this topic:
[[OpenPHDGuiding/phd2] Cannot compile PHD2 using latest INDI 2.0 (Issue #1032)](https://github.com/OpenPHDGuiding/phd2/issues/1032). I already provided there the link to the merged PR of @pawel-soja where the raw pointers are deprecated.

pawel-soja, 2023-02-02:
I've looked at the PHD2 sources and it looks like it's going to be a lot of work to get rid of the raw pointers. You can quickly do something like what I pasted below. It should compile with lots of 'deprecated' messages, but it should work.

https://github.com/OpenPHDGuiding/phd2/blob/472624a23bdd473630c0dd9575dc32b4c829845c/phdindiclient.h
```cpp
/*
 *  phdindiclient.h
 *  PHD2 Guiding
 *
 *  Created by Andy Galasso
 *  Copyright (c) 2018 Andy Galasso
 *  All rights reserved.
 *
 *  This source code is distributed under the following "BSD" license
 *  Redistribution and use in source and binary forms, with or without
 *  modification, are permitted provided that the following conditions are met:
 *    Redistributions of source code must retain the above copyright notice,
 *     this list of conditions and the following disclaimer.
 *    Redistributions in binary form must reproduce the above copyright notice,
 *     this list of conditions and the following disclaimer in the
 *     documentation and/or other materials provided with the distribution.
 *    Neither the name of Open PHD Guiding, openphdguiding.org, nor the names of its
 *     contributors may be used to endorse or promote products derived from
 *     this software without specific prior written permission.
 *
 *  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
 *  AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 *  IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
 *  ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
 *  LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
 *  CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
 *  SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
 *  INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
 *  CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
 *  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
 *  POSSIBILITY OF SUCH DAMAGE.
 *
 */
#ifndef PHDINDICLIENT_H
#define PHDINDICLIENT_H

#include <libindi/baseclient.h>
#include <libindi/basedevice.h>

class PhdIndiClient : public INDI::BaseClient
{
    bool m_disconnecting;

public:
    PhdIndiClient();
    ~PhdIndiClient();

    void serverDisconnected(int exit_code) final;

    virtual void IndiServerDisconnected(int exit_code) = 0;

    // must use this in PHD2 rather than BaseClient::disconnectServer()
    bool DisconnectIndiServer();

#if INDI_VERSION_MAJOR >= 2
public: // old deprecated interface INDI Version < 2.0.0
    virtual void newDevice(INDI::BaseDevice *dp) = 0;
    virtual void removeDevice(INDI::BaseDevice *dp) = 0;
    virtual void newProperty(INDI::Property *property) = 0;
    virtual void removeProperty(INDI::Property *property) = 0;

    virtual void newMessage(INDI::BaseDevice *dp, int messageID) = 0;
    virtual void newBLOB(IBLOB *bp) = 0;
    virtual void newSwitch(ISwitchVectorProperty *svp) = 0;
    virtual void newNumber(INumberVectorProperty *nvp) = 0;
    virtual void newText(ITextVectorProperty *tvp) = 0;
    virtual void newLight(ILightVectorProperty *lvp) = 0;

public: // new interface INDI Version >= 2.0.0
    void newDevice(INDI::BaseDevice device) override
    {
        return newDevice((INDI::BaseDevice *)device);
    }

    void removeDevice(INDI::BaseDevice device) override
    {
        return removeDevice((INDI::BaseDevice *)device);
    }

    void newProperty(INDI::Property property) override
    {
        return newProperty((INDI::Property *)property);
    }

    void removeProperty(INDI::Property property) override
    {
        return removeProperty((INDI::Property *)property);
    }

    void updateProperty(INDI::Property property) override
    {
        switch (property.getType())
        {
        case INDI_NUMBER: return newNumber((INumberVectorProperty *)property);
        case INDI_SWITCH: return newSwitch((ISwitchVectorProperty *)property);
        case INDI_LIGHT:  return newLight((ILightVectorProperty *)property);
        case INDI_BLOB:   return newBLOB((IBLOB *)INDI::PropertyBlob(property)[0].cast());
        case INDI_TEXT:   return newText((ITextVectorProperty *)property);
        }
    }

    void newMessage(INDI::BaseDevice device, int messageID) override
    {
        return newMessage((INDI::BaseDevice *)device, messageID);
    }
#endif
};

#endif
```

I'll try to write a migration guide to the new version over the weekend with examples as suggested by @knro.

paolostivanin, 2023-02-02:
Just to be clear, I am **not** pointing fingers or else here. I really do appreciate **a lot** all the work you do on INDI and KStars :smile: 
My observation/feedback/improvement suggestion is about release management. As a dev myself, I think it's good practice to clearly communicate breaking changes and, when possible, to provide example of migration as well.

(e.g.  like GNOME devs did for [GTK3->GTK4](https://docs.gtk.org/gtk4/migrating-3to4.html))

paolostivanin, 2023-02-02:
@pawel-soja thanks a lot :smile:

pawel-soja, 2023-02-07:
See https://github.com/OpenPHDGuiding/phd2/issues/1036
(I'm adding related links because the topic has grown into many threads.)

closed 2023-02-21 by knro