indilib/indi #1834: Lacerta MFOC Driver Hangs Up When Relative Positioning

timpeKH, 2023-02-02:
When I issue a relative position command in the INDI Control Panel, the Lacerta MFOC reacts with a
strange and wrong absolute position (huge, unrealistic number). Further the MFOC stops any reaction and communication with the INDI-Client. No disconnect or re-connection is possible unless the Lacerta MFOC handbox is switched off and restarted.

**Setup and Software Versions Used**
I use a Lacerta MFOC with firmware version 3.03. The MFOC is connected via USB to a Raspberry Pi 4, 8GB running Ubuntu Server arm64, 22.04 LTS (Jammy) as a INDI-host. So the Rapsberry Pi is where the indiserver runs. As clients EKOS or CCDCiel were used on a Laptop under XUbuntu 22.04. LTS ( KSTARS version 3.6.2 Stable Build 2022-11-30T18:14:22Z)
The driver package used on the RPI4 which provided the Lacerta MFOC driver (AFAIK) was/is : 
Package: libindi1
Version: 1.9.9+202211300913~ubuntu22.04.1

**To Reproduce**
1. Connect Lacerta MFOC Handbox with firmware 3.03 via USB and power on
2. Start indiserver and run driver the process for Lacerta MFOC: indi_lacerta_mfoc_focus along with a CCD simulator driver or any minimum configuration to use an indi client and the INDI Control Panel (My indiserver runs on a RPI4, 8GB with Ubuntu Server 22.04 LTS (Jammy). 
3. connect to indiserver via EKOS client (I connect via another Laptop to the RPI4) and start INDI Control Panel
4. In the INDI Control Panel go to the tab for the "Lacerta MFOC" and there to the "Main Control" tab. Press "connect" to connect to the Lacerta MFOC.
5. First set a "Absolute Position" value (e.g. 3000) and send the command by pressing the "set" button
7. Next set a "Relative Position" value (e.g. 1000) and send the command by pressing the "set" button
8. Check what happens to the feedback status value of "Absolute Position". In my case a huge integer number appears while the "Relative Position" command does not yield any focuser motion. Now the focuser should be completely non-reactive.
9. Try to shut down the indiserver process
10. Check if the Lacerta MFOC driver process "indi_lacerta_mfoc_focus" was stopped together with the indiserver. In my case this process keeps running until terminated by a kill signal.

If you could reproduce this issue you will probably need to switch off and on the Lacerta MFOC handbox before being able again to connect to the Lacerta MFOC from the INDI Control Panel.

**Expected behavior**
When executing step 7) the expected behaviour was that the focuser moves 1000 steps in the direction indicated by the "Direction" button. Assumed the absolute position is 3000 the focuser should move either to 2000 or 4000 steps.

**Screenshots**
see attachments
[image]

**Desktop (please complete the following information):**
 see above

**Log Files**
attached (from the RPI where the indiserver is running)

[indi_lacerta_mfoc_focus_21-33-16.log](https://github.com/indilib/indi/files/10573423/indi_lacerta_mfoc_focus_21-33-16.log)

pawel-soja, 2023-02-02:
4294967295 is 0xFFFFFFFF - looks like "-1"

timpeKH, 2023-02-02:
True, but that's not what I entered as relative position change. You can see the "-1" even in the log file. But I am not sure what it means. Could be an error return code from the MFOC(?)  (which was then taken as an absolute position by the client?)

chkettu, 2023-05-07:
merge request opened - should be fixed with next release!

closed 2024-04-06 by github-actions[bot]