indilib/indi #1836: indi-nut fails to compile - INDI v2.0.0

aaronwmorris, 2023-02-04:
indi-nut fails to compile under INDI v2.0.0.  (I do not use this server, I am just reporting the failure)

INDI:  v2.0.0
OS:  Debian 11
Platform: x86_64
libnutclient0:  2.7.4-13

```
Scanning dependencies of target indi_playerone_ccd
[ 94%] Building CXX object indi-playerone/CMakeFiles/indi_playerone_ccd.dir/playerone_ccd.cpp.o
[ 95%] Linking CXX executable indi_playerone_wheel
[ 95%] Built target indi_playerone_wheel
Scanning dependencies of target indi_nut
[ 96%] Building CXX object indi-nut/CMakeFiles/indi_nut.dir/indi_nut.cpp.o
/home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.cpp: In destructor ‘virtual NetworkUPSToolsMonitor::~NetworkUPSToolsMonitor()’:
/home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.cpp:46:5: warning: deleting object of polymorphic class type ‘nut::TcpClient’ which has non-virtual destructor might cause undefined behavior [-Wdelete-non-virtual-dtor]
   46 |     delete nutClient;
      |     ^~~~~~~~~~~~~~~~
/home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.cpp: In member function ‘virtual void NetworkUPSToolsMonitor::ISGetProperties(const char*)’:
/home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.cpp:94:21: error: ‘INDI::PropertyView<T>* INDI::PropertyBasic<T>::operator&() [with T = _IText]’ is protected within this context
   94 |     defineProperty(&nutMonitorUrl);
      |                     ^~~~~~~~~~~~~
In file included from /usr/local/include/libindi/indipropertytext.h:21,
                 from /usr/local/include/libindi/basedevice.h:29,
                 from /usr/local/include/libindi/parentdevice.h:3,
                 from /usr/local/include/libindi/defaultdevice.h:21,
                 from /usr/local/include/libindi/indiweather.h:27,
                 from /home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.h:29,
                 from /home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.cpp:26:
/usr/local/include/libindi/indipropertybasic.h:101:27: note: declared protected here
  101 |         PropertyView<T> * operator &();
      |                           ^~~~~~~~
/home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.cpp: In member function ‘virtual bool NetworkUPSToolsMonitor::updateProperties()’:
/home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.cpp:104:25: error: ‘INDI::PropertyView<T>* INDI::PropertyBasic<T>::operator&() [with T = _IText]’ is protected within this context
  104 |         defineProperty(&nutMonitorUrl);
      |                         ^~~~~~~~~~~~~
In file included from /usr/local/include/libindi/indipropertytext.h:21,
                 from /usr/local/include/libindi/basedevice.h:29,
                 from /usr/local/include/libindi/parentdevice.h:3,
                 from /usr/local/include/libindi/defaultdevice.h:21,
                 from /usr/local/include/libindi/indiweather.h:27,
                 from /home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.h:29,
                 from /home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.cpp:26:
/usr/local/include/libindi/indipropertybasic.h:101:27: note: declared protected here
  101 |         PropertyView<T> * operator &();
      |                           ^~~~~~~~
/home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.cpp: In member function ‘virtual bool NetworkUPSToolsMonitor::saveConfigItems(FILE*)’:
/home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.cpp:165:27: error: ‘INDI::PropertyView<T>* INDI::PropertyBasic<T>::operator&() [with T = _IText]’ is protected within this context
  165 |     IUSaveConfigText(fp, &nutMonitorUrl);
      |                           ^~~~~~~~~~~~~
In file included from /usr/local/include/libindi/indipropertytext.h:21,
                 from /usr/local/include/libindi/basedevice.h:29,
                 from /usr/local/include/libindi/parentdevice.h:3,
                 from /usr/local/include/libindi/defaultdevice.h:21,
                 from /usr/local/include/libindi/indiweather.h:27,
                 from /home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.h:29,
                 from /home/aaron/Projects/src/indi_3rdparty/indi-nut/indi_nut.cpp:26:
/usr/local/include/libindi/indipropertybasic.h:101:27: note: declared protected here
  101 |         PropertyView<T> * operator &();
      |                           ^~~~~~~~
make[2]: *** [indi-nut/CMakeFiles/indi_nut.dir/build.make:82: indi-nut/CMakeFiles/indi_nut.dir/indi_nut.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3625: indi-nut/CMakeFiles/indi_nut.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

aaronwmorris, 2023-02-04:
Sorry, wrong repo

closed 2023-02-04 by aaronwmorris

pawel-soja, 2023-02-04:
```cpp
defineProperty(&nutMonitorUrl);
```
Replace to:
```cpp
defineProperty(nutMonitorUrl);
```
it is compatible with previous indi version too.

aaronwmorris, 2023-02-04:
Opened under indi-3rdparty:  https://github.com/indilib/indi-3rdparty/issues/715

pawel-soja, 2023-02-04:
Okay, I'm going to the computer. It's hard to do anything on the phone. I haven't had the weather for 2 months, the new assembly is standing and waiting, I don't want to sleep, I'll pick something up.