indilib/indi #1838: Esatto2 Focuser - response parsing errors

aegiskeller, 2023-02-05:
I am using INDI in conjunction with astroberry distro (i.e. raspberrian) and I am getting a series of errors/warnings from INDI related to the use of the sesto senso 2 driver with my ESATTO2 device. I note that the ESATTO device is listed but "sesto senso 2" is the driver and indi_sestosenso2_focus is the executable.

The consequence is that I can drive the focuser unit to an absolute position but INDI does not know the resulting absolute position hence cannot use autofocus.

Perhaps the Esatto2 device has a different key-value set? For example, SPEED is now FOCUS_SPEED and ABS_POS is now ABS_FOCUS_POSITION

log excerpt:
[2023-02-05T15:26:57.381 AEDT DEBG ][     org.kde.kstars.ekos.focus] - Setting focus ticks to  272180
[2023-02-05T15:26:57.382 AEDT DEBG ][           org.kde.kstars.indi] - Esatto : "[DEBUG] Sending command: {\"req\":{\"cmd\":{\"MOT1\" :{\"GOTO\":272180}}}} with property: GOTO "
[2023-02-05T15:26:57.395 AEDT DEBG ][           org.kde.kstars.indi] - Esatto : "[DEBUG] Received response: {\"res\":{\"cmd\":{\"MOT1\":{\"GOTO\":\"done\"}}}} "
[2023-02-05T15:26:57.426 AEDT INFO ][           org.kde.kstars.indi] - Esatto :  "[INFO] Focuser is moving to position 272180 "
[2023-02-05T15:26:57.427 AEDT DEBG ][           org.kde.kstars.indi] - Esatto : "[DEBUG] Sending command: {\"req\":{\"get\":{\"MOT1\":\"\"}}} with property: SPEED "
[2023-02-05T15:26:57.472 AEDT DEBG ][           org.kde.kstars.indi] - Esatto : "[DEBUG] Received response: {\"res\":{\"get\":{\"MOT1\":{\"SUBMODEL\":\"ESATTO2\",\"NTC_T\":\"38.05\",\"BKLASH\":0,\"CAL_STATUS\":\"stop\",\"CAL_MAXPOS\":439000,\"CAL_BKLASH\":173,\"CAL_MIN_LO\":0,\"CAL_MIN_HI\":31325,\"CAL_MAX_LO\":399297,\"CAL_MAX_HI\":439000,\"CAL_OFFMIN\":31325,\"HSEND "
[2023-02-05T15:26:57.501 AEDT INFO ][           org.kde.kstars.indi] - Esatto :  "[ERROR] Failed to find property: SPEED "
[2023-02-05T15:26:57.531 AEDT INFO ][           org.kde.kstars.indi] - Esatto :  "[ERROR] Communication error: cmd {\"req\":{\"get\":{\"MOT1\":\"\"}}} property SPEED response:  "
[2023-02-05T15:26:57.972 AEDT DEBG ][           org.kde.kstars.indi] - Esatto : "[DEBUG] Sending command: {\"req\":{\"get\":{\"MOT1\":\"\"}}} with property: SPEED "
[2023-02-05T15:26:58.025 AEDT DEBG ][           org.kde.kstars.indi] - Esatto : "[DEBUG] Received response: {\"res\":{\"get\":{\"MOT1\":{\"SUBMODEL\":\"ESATTO2\",\"NTC_T\":\"38.05\",\"BKLASH\":0,\"CAL_STATUS\":\"stop\",\"CAL_MAXPOS\":439000,\"CAL_BKLASH\":173,\"CAL_MIN_LO\":0,\"CAL_MIN_HI\":31325,\"CAL_MAX_LO\":399297,\"CAL_MAX_HI\":439000,\"CAL_OFFMIN\":31325,\"HSEND "
[2023-02-05T15:26:58.081 AEDT INFO ][           org.kde.kstars.indi] - Esatto :  "[ERROR] Failed to find property: SPEED "
Before a movement command is issued:
[2023-02-05T16:13:12.148 AEDT DEBG ][           org.kde.kstars.indi] - Sesto Senso 2 : "[DEBUG] Sending command: {\"req\":{\"get\":{\"MOT1\":\"\"}}} with property: ABS_POS "
[2023-02-05T16:13:12.200 AEDT DEBG ][           org.kde.kstars.indi] - Sesto Senso 2 : "[DEBUG] Received response: {\"res\":{\"get\":{\"MOT1\":{\"SUBMODEL\":\"ESATTO2\",\"NTC_T\":\"38.36\",\"BKLASH\":0,\"CAL_STATUS\":\"stop\",\"CAL_MAXPOS\":439000,\"CAL_BKLASH\":173,\"CAL_MIN_LO\":0,\"CAL_MIN_HI\":31325,\"CAL_MAX_LO\":399297,\"CAL_MAX_HI\":439000,\"CAL_OFFMIN\":31325,\"HSEND "
[2023-02-05T16:13:12.201 AEDT INFO ][           org.kde.kstars.indi] - Sesto Senso 2 :  "[WARNING] Failed to process position response: _STEP280960 (11 bytes) "

knro, 2023-02-05:
Are both devices firmwares up to date? the latest driver only works with the latest firmware.

aegiskeller, 2023-02-06:
My bad - was using old astroberry indi driver. Rebuilt indi and works fine.

closed 2023-02-06 by aegiskeller