indilib/indi #1839: Client build failure on 20.04

knro, 2023-02-05:
KStars (at least) fails to [build on 20.04 ](https://launchpadlibrarian.net/648913949/buildlog_ubuntu-focal-amd64.kstars-bleeding_6%3A3.6.3+202301311034~ubuntu20.04.1_BUILDING.txt.gz) with an error about protected context. The offending function is this Qt signal:

void newINDIMessage(INDI::BaseDevice dp, int messageID);

On 22.04 and later, no such issues. Is this related to the GCC version @pawel-soja ?

pawel-soja, 2023-02-05:
sure, no problem. Where can I download this source version? I'm a bit confused, I usually download from https://github.com/KDE/kstars but I don't see a branch with changes for INDI Core 2.0.0. Can you help?

pawel-soja, 2023-02-05:
Okay I got it, I'm having trouble syncing.

pawel-soja, 2023-02-05:
Okay, I think I know what it is. There is an older `moc` generator on the mentioned system. Newer versions use the correct `std::addressof` function.
`Using Qt version 5.12.8`
```cpp
void ClientManager::newINDIMessage(INDI::BaseDevice _t1, int _t2)
{
    void *_a[] = { nullptr, const_cast<void*>(reinterpret_cast<const void*>(&_t1)), const_cast<void*>(reinterpret_cast<const void*>(&_t2)) };
    QMetaObject::activate(this, &staticMetaObject, 10, _a);
}
```

`Using Qt version 5.15.2`
```cpp
void ClientManager::newINDIMessage(INDI::BaseDevice _t1, int _t2)
{
    void *_a[] = { nullptr, const_cast<void*>(reinterpret_cast<const void*>(std::addressof(_t1))), const_cast<void*>(reinterpret_cast<const void*>(std::addressof(_t2))) };
    QMetaObject::activate(this, &staticMetaObject, 10, _a);
}
```

Protecting the address extraction, it was intended to catch errors in retrieving the address to temporary devices, where such behavior was already fixed in several places.
```cpp
class BaseDevice
{
/* ... */
    protected:
        BaseDevice *operator&()
        {
            return this;
        }
```

I don't think I see any other option than to remove this method - I'm still wondering because it's an interesting case :thinking:

knro, 2023-02-06:
So I presume if removed then it would not have any side-effects any both versions?

pawel-soja, 2023-02-06:
There should not be any problems https://github.com/indilib/indi/pull/1843

closed 2023-02-21 by knro