indilib/indi #1842: Driver "indi_lacerta_fmc_focus" (!) (for Firmware version 3.1): relative positioning inward causes out of range position

timpeKH, 2023-02-05:
When the driver "indi_lacerta_fmc_focus" (indilib1 2.0.0)  runs on a remote server and a client connects to it from another machine via network,
then a relative positioning command in inward direction ("focus in") causes a command (according to log: DEBUG	232.861839 sec	: CMD [: M -1 #]). This command is ignored by the device itself  and interestingly the Lacerta MFOC hand control panel is not affected. But the absolute position within the _INDI control panel_ gets wrong (4294967295 is 0xFFFFFFFF is -1). 
This yields an inability to command any other position unless the "sync" command is used to correct the wrong absolute position. 

_It is probably relevant to note that this issue does not occur when the device is connected locally and the device driver is run as a local service._

It is further of interest to note that the root cause of this bug is probably also responsible for another bug for another driver.
Pls. see also [https://github.com/indilib/indi/issues/1834](https://github.com/indilib/indi/issues/1834) for the driver "indi_lacerta_mfoc_focus". As opposed to this bug the driver "indi_lacerta_mfoc_focus" seems to crash completely due to the negative number in the position command. Further the firmware of the latter driver is not capable to re-connect until the hardware hand control panel was switched off and on again. 

**To Reproduce**
1. Run driver  "indi_lacerta_fmc_focus" on a remote server (e.g. Astroberry or Rasbberry/Ubuntu server running an indiserver)
2. Connect from a local client to the remote indiserver via KStars -> Tools -> Device -> Device Manager -> Client => external indiserver
3. Open INDI Contorl Panel
4. Enter relative position command as in the following screenshot (e.g. 500 steps or any other sensible number without making the hardware running into motion limits):
[image]
5. The result should look as on the following screenshot - **note the absolute position (!)**:
[image]
 

**Expected behavior**
Correct inward position movement of focuser device

**Desktop (please complete the following information):**
 - Host running indiserver process: Raspberry Pi4, 8GB, Ubuntu Server 22.04 LTS arm64, indilib1 2.0.0
 - Client: Tuxedo Laptop XUbuntu 22.04 LTS amd64 with KStars Build 2023-01-31T10:34:48Z

Although there is not much to see I attach a log file. Please note the line 
"DEBUG	63.338030 sec	: CMD [: M -1 #]"
which seems to be the consequence of the relative positioning command inward .
[RPI4_Server_indi_lacerta_mfoc_fmc_focus_16-45-04.log](https://github.com/indilib/indi/files/10611812/RPI4_Server_indi_lacerta_mfoc_fmc_focus_16-45-04.log)

knro, 2023-02-06:
@chkettu Can you please take a look? was this introduced in #1826 ?

chkettu, 2023-02-06:
Yes, I'll take a look - but can't compile/test until end of february (due to health rehabilitation stay).

timpeKH, 2023-02-11:
@chkettu: take all the time you need - your health is more important. This is a hobby instrument and I really appreciate if you have look when you're doing well again! Although I did not compile the indi libraries by myself so far (used the convenient Ubuntu packages) I am willing to assist with testing if this can be of any help. I could also learn how to compile the libs from source if needed.

chkettu, 2023-05-07:
merge request opened - should be fixed with next release!

timpeKH, 2023-06-04:
@chkettu: Thanks for looking into this issue. I tested it. The previous behaviour has stopped. But the *relative* positioning does not work correctly yet: The "Direction" button in the INDI Control Panel is not recognized correctly. When moving in direction "Focus In", the focuser moves endless in "Focus Out" direction. The number of steps is ignored and the motion ends in the hard stops where the stepper motor then hangs. This behaviour is repeatable. The relative positioning works correctly only in "Focus Out" direction.

timpeKH, 2023-06-04:
P.S.: I tested with the release from 31. may 2023.

chkettu, 2023-06-05:
I'm going to check with latest release.

singraber, 2023-08-21:
I recently purchased a Lacerta MFOC with firmware 3.1.01 and found a similar behavior. When attempting to make a relative position change in the "Focus In" direction, the hand controller will refuse and alert that the desired position is above the maximum value (actually it should be decreased when moving "in"). I had a look at the driver code and found a problematic line which is responsible for the calculation of the new position. I was able to write a fix, I can now correctly move in and out with relative changes. For more details see #1924.

closed 2024-04-06 by github-actions[bot]