indilib/indi #1846: Sesto Senso 2 Focuser will not move.

setnes, 2023-02-10:
I can't move focuser.  Here's the relevant error from the log.

```
ERROR	401.087449 sec	: Required {"req":{"cmd":{"MOT1":{"MOVE_ABS":{"STEP":500}}}}} failed: {"res":{"cmd":{"MOT1":{"MOVE_ABS":"Error: invalid command"}}}}
ERROR	401.087652 sec	: Focuser failed to move to new requested position.
```

Using INDI stable Version: 2.0.0+202301310940~ubuntu22.04.1
indi_sestosenso2_focus driver
OS:  Armbian (Ubuntu Jammy)

knro, 2023-02-10:
Are you using latest firmware release from Primaluce?

setnes, 2023-02-10:
Great idea.  I am not on the latest firmware.  I didn't consider that.  This device was working on Astroberry before I tried on this new hardware.  I will upgrade the firmware and try this tonight.  You are probably right though.  

I just need to find a windows computer to do the firmware upgrade.  <sigh>

setnes, 2023-02-10:
The firmware update fixed the problem, but it created another one for me.  The focuser works on the new computer I am building, but not on Astroberry.  I was planning on using Astroberry tonight.  That's stuck with an older INDI.   Yuck.

I'll close this ticket...

closed 2023-02-10 by setnes