indilib/indi #1848: FITS Error: bad float to string conversion

sterne-jaeger, 2023-02-13:
Camera drivers run into problems if the FITS header SCALE is not a number but it is "inf" (e.g. in the case that the scope focal length is 0 since it is unknown). In this case, indiccd.cpp::2304 reports the error

 **FITS Error: bad float to string conversion**

**To Reproduce**
Exact steps to reproduce the behavior.
1. Start the INDI server with a camera driver and a telescope driver
2. Ensure that the focal length is 0 in the telescope driver
3. Capture an image
4. This should produce the above error message, FITS file will not be created.

**Expected behavior**
Skip FITS header and store the FITS file.

**Desktop (please complete the following information):**
 - OS: Raspberry PI OS 11 (bullseye)
 - INDI 2.0.1

nouspiro, 2023-02-16:
Should be fixed in #1847

knro, 2023-02-17:
@sterne-jaeger Can you confirm fix?

knro, 2023-02-21:
Will close this as fixed. @sterne-jaeger please reopen if the issue is still there.

closed 2023-02-21 by knro