indilib/indi #1849: Compilation blocked at 60%

HamHamFonFon, 2023-02-14:
During compilation, cmake is blocking at building "indi_dmfc_focus" driver.
Last output:
```
[ 59%] Built target indi_lakeside_focus
[ 59%] Building CXX object drivers/focuser/CMakeFiles/indi_dmfc_focus.dir/dmfc.cpp.o
[ 60%] Building CXX object drivers/focuser/CMakeFiles/indi_esattoarco_focus.dir/primalucacommandset.cpp.o
[ 60%] Linking CXX executable indi_dmfc_focus
[ 60%] Built target indi_dmfc_focus
```
No error output, no verbosity...i cant tell more. It's blocked on it since 2 hours.

**Desktop :**
 - OS: Ubuntu
 - Version 22.04
- Support : Rasbperry PI 3 Model B+

knro, 2023-02-21:
This is an issue with compiling on raspberry pi, not an issue with INDI itself.

closed 2023-02-21 by knro

HamHamFonFon, 2023-02-21:
Just before close definitvely, is because not enough RAM for compiling (only 1Gb) or cmake version too old (version 23) ?