indilib/indi #1861: EAF autofocus error in EKOS

VRzichtbaar, 2023-02-28:
**Describe the bug**
The EAF focusser can be uses manually and works as expected. Once autofocus is used a failure to read temperature error is caused. The mound driver crashes and the system does not recover when stopping the autofocus request

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver EAF
2. Perform action start quro focus
3. See error temperature reading error

**Expected behavior**
to execute the autofocus script

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: astroberry
 - Version [e.g. 1.7.6, GIT..etc]

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html
[log_16-20-59.txt](https://github.com/indilib/indi/files/10852184/log_16-20-59.txt)

knro, 2023-03-01:
In the log I see this:

```
[2023-02-28T16:24:11.501 CET INFO ][           org.kde.kstars.indi] - ASI EAF :  "[ERROR] Failed to read position. Error: 4 "
[2023-02-28T16:24:11.503 CET INFO ][           org.kde.kstars.indi] - ASI EAF :  "[ERROR] Failed to read temperature. Error: 4 "
[2023-02-28T16:24:12.003 CET INFO ][           org.kde.kstars.indi] - ASI EAF :  "[ERROR] Failed to read position. Error: 4 "
[2023-02-28T16:24:12.006 CET INFO ][           org.kde.kstars.indi] - ASI EAF :  "[ERROR] Failed to read temperature. Error: 4 "
```
So both position and temperature cannot be read. and this happens when focuser is in motion only?

VRzichtbaar, 2023-03-01:
The strange thing is that when I use the focusser via the buttons (in-out-goto position) everything works fine.
I even get the V curve when refreshing the image and setting the auto star selection.
The temperature reading from the EAF as seen in the focusser tab seems perfectly fine. (updating and accurate) 
Only when I press auto focus all the errors occur. The position reading from the mount works fine in every other function.

When I stop the autofocus action the system does not recover. The temperature errors keep coming and this can only be resolved by completely stopping and restarting EKOS and Kstars.

VRzichtbaar, 2023-03-01:
Just updated the firmware from 3.3.3 to 3.3.7 and the system does not crash on the error anymore.
I still have to test with some stars this evening but this looks hopefull.
I hope I did not waste anyones time.

VRzichtbaar, 2023-03-01:
Indeed the firmware version 3.3.3 of the ZWO EAF 5v focusser caused the issue. 
After upgrading to 3.3.7 everything worked as expected.
knro really thanks for looking at my issue report.

closed 2023-03-02 by knro