indilib/indi #1878: indi_canon_ccd crashes after 30 frames on Astroberry when remote indi web server used

mikko, 2023-03-26:
**Describe the bug**
When I'm shooting a sequence using KStars, after around 30 frames, the indi_canon_ccd driver child process dies on the Raspberry running Astroberry. Looks like the driver keeps reserving more and more memory until it crashes. On the client side, on laptop, it looks like Kstars is crashing, but really it's the driver behind the web server.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Start KStars with remote INDI connection to Astroberry with Canon EOS 60D connected to it using USB
2. Start a long imaging sequence of, for example 100 frames
3. Wait for the sequence to proceed further until INDI Control Panel pops up on KStars after maybe 30 frames (on my RPi)
4. Witness the sequence stopping (and KStars crashing after restarting or stopping the sequence, but I guess this is another bug)

**Expected behavior**
The whole imaging sequence should execute without problems.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Client Ubuntu 20.04 Kstars 3.6.3 (Tested also on Windows 10). Server Raspberry 3B+ running Astroberry 2.0.4
 - Version [e.g. 1.7.6, GIT..etc]

**Log Files**

Memory usage of the indi_canon_ccd child process looks like this:
```bash
astroberry@astroberry:~ $ while true; do ps -p 2670 -o rss=; sleep 2; done;
19604
19864
19864
28032
74960
#middle part omitted
661064
644088
676428
676704
678112
#client dead
```
KStars client log

```
[2023-03-26T18:24:26.199 EEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS 60D : "[DEBUG] Downloading image... Name: (capt0071.cr2) Folder: (/) Delete from SD card? (false) "
[2023-03-26T18:24:31.687 EEST DEBG ][           org.kde.kstars.indi] - Removing device "Canon DSLR EOS 60D"
[2023-03-26T18:24:31.689 EEST DEBG ][           org.kde.kstars.indi] - INDIListener: Removing device "Canon DSLR EOS 60D"
[2023-03-26T18:24:31.690 EEST INFO ][           org.kde.kstars.ekos] - "Canon DSLR EOS 60D is offline."
[2023-03-26T18:24:34.599 EEST DEBG ][           org.kde.kstars.indi] - Received new device Canon DSLR EOS 60D
[2023-03-26T18:24:34.606 EEST DEBG ][           org.kde.kstars.indi] - INDIListener: New device "Canon DSLR EOS 60D"
[2023-03-26T18:24:34.606 EEST INFO ][           org.kde.kstars.ekos] - Ekos received a new device:  "Canon DSLR EOS 60D"
[2023-03-26T18:24:34.607 EEST DEBG ][ org.kde.kstars.ekos.scheduler] - Scheduler INDI status is 0
[2023-03-26T18:24:34.607 EEST DEBG ][ org.kde.kstars.ekos.scheduler] - Scheduler Ekos status is 2
[2023-03-26T18:24:34.620 EEST DEBG ][           org.kde.kstars.indi] - < Canon DSLR EOS 60D >: < CONNECTION >
```

```bash
$ cat /tmp/indiserver.log
2023-03-26T14:58:57: startup: indiserver -p 7624 -m 100 -v -f /tmp/indiFIFO 
2023-03-26T14:58:57: listening to port 7624 on fd 3
FIFO: start indi_canon_ccd
FIFO: Starting driver indi_canon_ccd
2023-03-26T14:58:57: Driver indi_canon_ccd: pid=2670 rfd=4 wfd=7 efd=8
FIFO: start indi_lx200gps
FIFO: Starting driver indi_lx200gps
2023-03-26T14:58:57: Driver indi_lx200gps: pid=2672 rfd=5 wfd=10 efd=11
2023-03-26T14:58:57: Driver indi_lx200gps: initializing from LX200 GPS device...
2023-03-26T14:58:57: Driver indi_lx200gps: Initializing from Generic LX200 device...
2023-03-26T14:58:57: Driver indi_lx200gps: snooping on GPS Simulator.GEOGRAPHIC_COORD
2023-03-26T14:58:57: Driver indi_lx200gps: snooping on GPS Simulator.TIME_UTC
2023-03-26T14:58:57: Driver indi_lx200gps: snooping on Dome Simulator.DOME_PARK
2023-03-26T14:58:57: Driver indi_lx200gps: snooping on Dome Simulator.DOME_SHUTTER
2023-03-26T14:58:57: Driver indi_canon_ccd: Number of cameras detected: 1.
2023-03-26T14:58:57: Driver indi_canon_ccd: Detected camera model Canon EOS 60D on port usb:001,007
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on Telescope Simulator.TELESCOPE_INFO
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on Telescope Simulator.GEOGRAPHIC_COORD
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on Focuser Simulator.ABS_FOCUS_POSITION
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on Focuser Simulator.FOCUS_TEMPERATURE
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on CCD Simulator.FILTER_SLOT
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on CCD Simulator.FILTER_NAME
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on SQM.SKY_QUALITY
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on LX200 GPS.EQUATORIAL_EOD_COORD
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on LX200 GPS.TELESCOPE_INFO
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on LX200 GPS.GEOGRAPHIC_COORD
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on Canon DSLR EOS 60D.ABS_FOCUS_POSITION
2023-03-26T14:58:57: Driver indi_canon_ccd: snooping on Canon DSLR EOS 60D.FOCUS_TEMPERATURE
2023-03-26T15:17:36: Client 6: new arrival from 192.168.1.235:42548 - welcome!
2023-03-26T15:17:37: Client 9: new arrival from 192.168.1.235:42550 - welcome!
Child process 2670 died
2023-03-26T15:24:31: Driver indi_canon_ccd: stderr EOF
<delProperty device="Canon DSLR EOS 60D"/>
2023-03-26T15:24:31: Driver indi_canon_ccd: restart #1
2023-03-26T15:24:31: Driver indi_canon_ccd: pid=3045 rfd=4 wfd=12 efd=13
2023-03-26T15:24:34: Driver indi_canon_ccd: Number of cameras detected: 1.
```

knro, 2023-03-26:
This indeed looks like a memory leak, which INDI version is used?

mikko, 2023-03-27:
On the raspberry (where the driver crashes):
```
INDI Library: 1.9.2
Code v1.9.2. Protocol 1.7.

```

On the client laptop:
```
INDI Library: 2.0.0
Code 2.0.0-tgz. Protocol 1.7.
```

Now that I realize the difference, I could upgrade INDI on the raspberry. Since it's the default Astroberry image and I didn't install it myself, I have to figure out how to do the upgrade first.

knro, 2024-03-03:
Can you try updating Raspberry PI to latest INDI and test again?

closed 2024-05-10 by github-actions[bot]