indilib/indi #1882: BaseClientQt example implementation

rickbassham, 2023-04-03:
I'm looking for an example on using the BaseClientQt class.

1. When installing `libindi-dev`, I can't find the necessary include file.
2. When installing from source with the following command, I can find the include file, but get an error when linking.

```
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Release -DINDI_BUILD_QT5_CLIENT=ON ../indi
make
make install
```

```
[main] Building folder: indi-qt-test 
[build] Starting build
[proc] Executing command: /usr/bin/cmake --build /workspaces/indi-qt-test/build --config Debug --target all -j 8 --
[build] [ 20%] Automatic MOC for target indi-qt-test
[build] [ 20%] Built target indi-qt-test_autogen
[build] Consolidate compiler generated dependencies of target indi-qt-test
[build] [ 40%] Linking CXX executable indi-qt-test
[build] /usr/bin/ld: CMakeFiles/indi-qt-test.dir/indi-qt-test_autogen/mocs_compilation.cpp.o: in function `MyClient::qt_metacast(char const*)':
[build] /workspaces/indi-qt-test/build/indi-qt-test_autogen/EWIEGA46WW/moc_client.cpp:86: undefined reference to `INDI::BaseClientQt::qt_metacast(char const*)'
[build] /usr/bin/ld: CMakeFiles/indi-qt-test.dir/indi-qt-test_autogen/mocs_compilation.cpp.o: in function `MyClient::qt_metacall(QMetaObject::Call, int, void**)':
[build] /workspaces/indi-qt-test/build/indi-qt-test_autogen/EWIEGA46WW/moc_client.cpp:91: undefined reference to `INDI::BaseClientQt::qt_metacall(QMetaObject::Call, int, void**)'
[build] /usr/bin/ld: CMakeFiles/indi-qt-test.dir/indi-qt-test_autogen/mocs_compilation.cpp.o:(.data.rel.ro+0x0): undefined reference to `INDI::BaseClientQt::staticMetaObject'
[build] /usr/bin/ld: CMakeFiles/indi-qt-test.dir/indi-qt-test_autogen/mocs_compilation.cpp.o:(.data.rel.ro._ZTI8MyClient[_ZTI8MyClient]+0x10): undefined reference to `typeinfo for INDI::BaseClientQt'
[build] /usr/bin/ld: /usr/lib/aarch64-linux-gnu/libindiclientqt.so: undefined reference to `vtable for INDI::BaseClientQt'
[build] collect2: error: ld returned 1 exit status
[build] gmake[2]: *** [CMakeFiles/indi-qt-test.dir/build.make:131: indi-qt-test] Error 1
[build] gmake[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/indi-qt-test.dir/all] Error 2
[build] gmake: *** [Makefile:91: all] Error 2
[proc] The command: /usr/bin/cmake --build /workspaces/indi-qt-test/build --config Debug --target all -j 8 -- exited with code: 2
[build] Build finished with exit code 2
```

Does anyone make use of this class, and can you point me to what I'm doing wrong here? Code I'm using to test this is located here: https://github.com/rickbassham/indi-qt-test

I'm new to QT programming, but not new to C++, so I may be just missing something easy.

knro, 2023-04-04:
On which OS is this? I can build this fine in GIT without any errors.

rickbassham, 2023-04-04:
I'm running it in a docker container running ubuntu. Relevant dockerfile: https://github.com/rickbassham/indi-qt-test/blob/main/.devcontainer/Dockerfile

pawel-soja, 2023-06-23:
Hello everyone!

I think it's a function missing in the compiled library that the Meta-Object Compiler creates.

There are two solutions, compiling the library with MOC, or adding header files to the final project.
1. In the first case, I can post a pull request and make it available for testing - @knro  decision.

2. The second solution is to add a path with a header like below. Sorry for the direct file path - I'm just showing where the problem is
```cmake
add_executable(${PROJECT_NAME} main.cpp client.cpp /usr/include/libindi/baseclientqt.h)
```

knro, 2023-06-24:
@pawel-soja Thank you. Go ahead with the MOC PR, I take this for granted in KStars I forgot about it here!

luzpaz, 2024-01-05:
What's left to do here ?

closed 2024-04-05 by github-actions[bot]