indilib/indi #1883: indi 2.0.1 compilation issue

paolostivanin, 2023-04-04:
Hello,
compilation of INDI 2.0.1 failed on openSUSE Tumbleweed with:

```
[   27s] /usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: /tmp/cc7mLd6s.ltrans0.ltrans.o: in function `check_lx200_connection(int)':
[   27s] /home/abuild/rpmbuild/BUILD/indi-2.0.1/drivers/telescope/lx200driver.cpp:72: undefined reference to `INDI::Logger::getInstance()'
[   27s] /usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: /home/abuild/rpmbuild/BUILD/indi-2.0.1/drivers/telescope/lx200driver.cpp:72: undefined reference to `INDI::Logger::print(char const*, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, char const*, ...)'
[   27s] /usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: /home/abuild/rpmbuild/BUILD/indi-2.0.1/drivers/telescope/lx200driver.cpp:91: undefined reference to `tty_read'
[   27s] /usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: /home/abuild/rpmbuild/BUILD/indi-2.0.1/drivers/telescope/lx200driver.cpp:100: undefined reference to `INDI::Logger::getInstance()'
[   27s] /usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: /home/abuild/rpmbuild/BUILD/indi-2.0.1/drivers/telescope/lx200driver.cpp:100: undefined reference to `INDI::Logger::print(char const*, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, char const*, ...)'
[   27s] /usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: /home/abuild/rpmbuild/BUILD/indi-2.0.1/drivers/telescope/lx200driver.cpp:94: undefined reference to `INDI::Logger::getInstance()'
[   27s] /usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: /home/abuild/rpmbuild/BUILD/indi-2.0.1/drivers/telescope/lx200driver.cpp:94: undefined reference to `INDI::Logger::print(char const*, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, char const*, ...)'
[   27s] /usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: /tmp/cc7mLd6s.ltrans0.ltrans.o: in function `LX200Telescope::checkConnection() [clone .localalias]':
[   27s] /home/abuild/rpmbuild/BUILD/indi-2.0.1/drivers/telescope/lx200telescope.cpp:264: undefined reference to `INDI::DefaultDevice::isSimulation() const'
[   27s] /usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: /tmp/cc7mLd6s.ltrans0.ltrans.o: in function `LX200Telescope::Handshake()':
[   27s] /home/abuild/rpmbuild/BUILD/indi-2.0.1/drivers/telescope/lx200telescope.cpp:264: undefined reference to `INDI::DefaultDevice::isSimulation() const'
[   27s] /usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: /tmp/cc7mLd6s.ltrans0.ltrans.o: in function `ACK(int)':
[   27s] /home/abuild/rpmbuild/BUILD/indi-2.0.1/drivers/telescope/lx200driver.cpp:110: undefined reference to `INDI::Logger::getInstance()'
...
```

v2.0.0 compiles fine on all archs.

xsnrg, 2023-04-04:
The builds for all the versions of Fedora seem to be working. If I look for lx200driver.cpp in this logfile: https://download.copr.fedorainfracloud.org/results/xsnrg/libindi-bleeding/fedora-37-x86_64/05739973-indi/builder-live.log.gz , this is the result:
```[ 14%] Building CXX object drivers/telescope/CMakeFiles/indilx200.dir/lx200driver.cpp.o
cd /builddir/build/BUILD/indi-master/drivers/telescope && /usr/bin/g++ -DENABLE_INDI_SHARED_MEMORY -DGIT_TAG_STRING=\"2.0.1-tgz\" -DHAVE_CLOCK_GETTIME -DHAVE_LIBNOVA -DHAVE_TIMESPEC_GET -DWITH_ENCLEN -D_XOPEN_SOURCE=700 -Dindilx200_EXPORTS -I/builddir/build/BUILD/indi-master/drivers/telescope -I/builddir/build/BUILD/indi-master/libs -I/usr/include/cfitsio -I/usr/include/libusb-1.0 -I/builddir/build/BUILD/indi-master/libs/indibase -I/builddir/build/BUILD/indi-master/libs/indibase/timer -I/builddir/build/BUILD/indi-master/libs/indidevice/. -I/builddir/build/BUILD/indi-master/libs/indidevice/property -I/builddir/build/BUILD/indi-master/libs/indidevice/../.. -I/builddir/build/BUILD/indi-master/libs/indicore/. -I/builddir/build/BUILD/indi-master/libs/indicore/../.. -I/builddir/build/BUILD/indi-master/libs/indicore -O2  -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_FORTIFY_SOURCE=2 -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-format-truncation -g -DHAVE_MREMAP -std=gnu++17 -fPIC -fPIC -MD -MT drivers/telescope/CMakeFiles/indilx200.dir/lx200driver.cpp.o -MF CMakeFiles/indilx200.dir/lx200driver.cpp.o.d -o CMakeFiles/indilx200.dir/lx200driver.cpp.o -c /builddir/build/BUILD/indi-master/drivers/telescope/lx200driver.cpp```

The log snippet above indicates a problem with linking where elements are not found in lx200driver... Are there any additional errors or warnings around the build of `lx200driver`? Also, check if an older version of the INDI lib is installed elsewhere on the system. If the LD_LIBRARY_PATH is looking in the system first, it could just be linking against the wrong version.

paolostivanin, 2023-04-04:
Hello,
OBS passes the following cmake opt: `-DCMAKE_SHARED_LINKER_FLAGS=" -Wl,--as-needed -Wl,--no-undefined -Wl,-z,now"` .
Removing the `--no-undefined` fixed the issue.

closed 2023-04-04 by paolostivanin