indilib/indi #1886: Omegon 571c driver crash

archcam, 2023-04-05:
I get this error with my Omegon 571c:

[2023-04-05T18:13:30.969 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-05T16:13:30: Driver indi_omegonprocam_ccd: pid=11937 rfd=17 wfd=17 efd=18"
[2023-04-05T18:13:30.969 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2023-04-05T18:13:30.978 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-05T16:13:30: Driver indi_omegonprocam_ccd: indi_omegonprocam_ccd: symbol lookup error: indi_omegonprocam_ccd: undefined symbol: _ZN4INDI3CCD15addFITSKeywordsEPNS_7CCDChipE"
[2023-04-05T18:13:30.978 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2023-04-05T18:13:30.979 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-05T16:13:30: Driver indi_omegonprocam_ccd: read: Connection reset by peer"
[2023-04-05T18:13:30.979 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-05T16:13:30: Driver indi_omegonprocam_ccd: Terminated after #0 restarts."
[2023-04-05T18:13:30.979 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""

archcam, 2023-04-05:
The driver stops immediately without initiating the connection with the camera. I tried both ToupTek drivers and Omegon drivers but the result is the same.
[log_23-34-11.zip](https://github.com/indilib/indi/files/11163315/log_23-34-11.zip)

knro, 2023-04-05:
How was this installed?

archcam, 2023-04-06:
I followed the guide on the indilib site for the ubuntu system

knro, 2023-04-06:
The packages were getting updated at that time, can you check now?

archcam, 2023-04-06:
I reinstalled Ubuntu and immediately installed indi and kstars according to the site's instructions without doing anything else... but the problem remains... I attach the log files.
[log_20-13-03 DRIVER OMEGON PRO.zip](https://github.com/indilib/indi/files/11173457/log_20-13-03.DRIVER.OMEGON.PRO.zip)
[log_20-15-39 DRIVER OMEGON EXE TOUPCAM.zip](https://github.com/indilib/indi/files/11173459/log_20-15-39.DRIVER.OMEGON.EXE.TOUPCAM.zip)
[log_20-17-21 DRIVER TOUPCAM.zip](https://github.com/indilib/indi/files/11173460/log_20-17-21.DRIVER.TOUPCAM.zip)

archcam, 2023-04-08:
I fixed it: after installing ubuntu and indilib/kstars I updated the software via terminal with the apt-get command, but the error and the driver crash remained. I then used the software update app that came with ubuntu and finally now everything works...
I thought updating via terminal with the apt-get command was low-level... ;-)

closed 2023-04-08 by archcam

knro, 2023-04-10:
Are you Ubuntu or Kubuntu?

archcam, 2023-04-10:
Ubuntu 22.04 lts