indilib/indi #1890: INDIServer gets the port number wrong when starting.

T-STUDIO, 2023-04-26:
The 2.01 version of INDIServer will start on a different port number than configured.

The error log is below

2023-04-27T00:26:02 INDI services stopped.

2023-04-27T00:26:02 Lost connection to local INDI server localhost:7,624
2023-04-27T00:26:01 INDI services started on port 7,624. Please connect devices.
2023-04-27T00:26:00 Starting INDI services...

Same result if I change the server start method like Ekos, INDIWeb Manager, INDI Starter.

I can't use it because the server won't start.

The environment is
Ubuntu 22.04 64-bit

Language environment Japanese (no change even if changed to English)

knro, 2023-06-01:
If you run this:

indiserver -v -p 8000 indi_simulator_ccd

Does it say it starts on 8000? by default it should start on 7624

d33psky, 2023-10-05:
@T-STUDIO when this happens, can you show the output of these commands : 
```
sudo netstat -nalp | egrep 'LISTEN .*indi'

ps axfwww | grep -C9 indiserver
```

So that we can hopefully see what is going on.

closed 2024-01-17 by knro