indilib/indi #1899: The optical train implementation also affects driver connectivity and server startup.

T-STUDIO, 2023-05-17:
A very problematic bug. It is a function added to Ekos, but it seems that it is also implemented on the INDI side, so I will report it.

The optical train implementation also affects driver connectivity and server startup.
Drivers (AUX drivers) that cannot be selected in the optical train category (camera, focuser, etc.) cannot be registered in the optical train.

Also, drivers who cannot register with the optical train will not be able to connect from the control panel.

There was a connection failure in this update, but there was a driver that can be avoided by connecting from CCDCiel.
However, if the driver is not in the optical train item, it cannot be selected even in CCDCiel, so the problem cannot be solved.

I can't use induno-simpleSwicher on my device.

If you update your OS and update INDI to the version that added the optical train like I did, you won't be able to connect to the server with the old settings.

Since the optical train was implemented without improving the Ekos server startup and connection rules that I pointed out earlier, there is a problem that the connection will be inconsistent and the driver and server will not start.

If there is a device that cannot be selected in the optical train item, please try it.

You should be able to start the driver, but you should not be able to connect.

knro, 2023-05-17:
Please tell us how to reproduce this issue exactly step by step.

knro, 2023-05-17:
I'm sorry but I need specific steps to see what the issue is. Steps like this:

1. Create profile in CCDCiel with these drivers.
2. Start profile in CCDCiel
3. X or Y does not work

Something like this.

T-STUDIO, 2023-05-17:
did you read the article?

It's all written.

T-STUDIO, 2023-05-17:
Problems can occur with any driver that boots with Ekos.

This is because when Ekos is started, the optical train automatically starts and wraps the driver.

Some AUX drivers are not selectable in the optical train. (Because there is no item) In this case the driver will not be able to connect even if it starts.

I have reported the trouble many times in the article on the forum and submitted the log.

are you reading? ? ?

gnthibault, 2023-07-31:
I am experiencing a complete failure of the indi_duino driver, although I checkd 100 times everything, I can never connect the driver from ekos indi client. Could that be the issue ?

Thank you in advance for your help, let me know if there are additional information I can provide

FalCT60, 2023-08-20:
Running stellarmate on RPI4B 8GB.
I am experiencing the same bug after updating my system.
Reflashing as of before august it all works fine, but as soon as the updates are applied all go wrong again.
Attached are indi and kstars logs.
[indi_app_log_23-08-20T09-49-09.txt](https://github.com/indilib/indi/files/12387647/indi_app_log_23-08-20T09-49-09.txt)
[indi_debug_log_23-08-20T09-49-09.txt](https://github.com/indilib/indi/files/12387648/indi_debug_log_23-08-20T09-49-09.txt)
[kstars_debug_log_2023-08-20T09-48-21.txt](https://github.com/indilib/indi/files/12387649/kstars_debug_log_2023-08-20T09-48-21.txt)
[log_09-48-21.txt](https://github.com/indilib/indi/files/12387650/log_09-48-21.txt)
[log_09-49-59.txt](https://github.com/indilib/indi/files/12387651/log_09-49-59.txt)
[log_09-54-17.txt](https://github.com/indilib/indi/files/12387652/log_09-54-17.txt)

knro, 2023-08-20:
This is not related to Optical trains as it has nothing to do with INDI drivers. @FalCT60 I see ZWO driver failing to startup. In your equipment profile, why do you have your profile set to remote and then remote host is localhost? any reason for this? It should be LOCAL and not remote.

FalCT60, 2023-08-20:
I followed [the advice from Jasem](https://indilib.org/forum/stellarmate/13668-ekos-indi-setup-help-needed.html#94726), to work around [a problem I am facing](https://indilib.org/forum/stellarmate/13668-ekos-indi-setup-help-needed.html) since the beginning.
Anyway, even if I revert to Local, [it will not work either](https://indilib.org/forum/stellarmate/13707-indi-perpetual-loop.html).
And I am in a situation I cannot use it anymore.
If I delete all indi & kstars configuration files and folders, then restart the app and re-create all my equipments and local trains it will work once, but then it will hang again.
I lost my whole afternoon and evenig yesterday, and now it seems it goes the same way.
If I reflash the sd card with the image I downloaded on 10th august it works, but as soon as I update it does not anymore.
[log_10-32-24.txt](https://github.com/indilib/indi/files/12387708/log_10-32-24.txt)
[log_10-34-57.txt](https://github.com/indilib/indi/files/12387709/log_10-34-57.txt)

knro, 2023-08-20:
The ZWO driver is crashing, please submit a ticket with StellarMate.

FalCT60, 2023-08-20:
OK, sorry, I thought this was the same bug.

luzpaz, 2023-12-17:
@FalCT60 can you link to that bug when you do ?

FalCT60, 2023-12-17:
Hello @luzpaz,
As far as I remember, my problem was not related to this bug.
Rgrds,

closed 2024-01-01 by knro