indilib/indi #1908: Non-Guide Dither Pulse causes Exposure timeout

fenriques, 2023-06-16:
When using Non-Guide Dither Pulse on a 10Micron mount (LX200 10micron), the camera (ASI2600) experiences a timeout.

Exact steps to reproduce the behavior.
1. Start Kstars/Ekos with LX200 10micron, indi_asi_ccd drivers
2. Execute a 10x30s sequence with any filter and focus position and  select 'non guide dither pulse' in the Guide Options tab. See attached image for exact parameters.
3. After dithering, the camera doesn't resume exposure and eventually the driver crashes. See the attached log.

It seems that after dithering, the guide system tries to resume a guide session taking a 2s image. But it is not needed as it's unguided.

Warning: in the log the mount is parked (the obs is closed) but it doesn't affect the result.

Relevant events in the log:
[2023-06-16T09:27:47.685 CEST DEBG ][           org.kde.kstars.indi] - Image received. Mode: "Normal" Size: 52191360
[2023-06-16T09:27:47.700 CEST INFO ][           org.kde.kstars.indi] - "FITS" file saved to "/home/ferrante/Light/Luminance/Light_Luminance_30_secs_2023-06-16T09-27-47_005.fits"
[2023-06-16T09:27:47.769 CEST INFO ][   org.kde.kstars.ekos.capture] - "Download Time: 0.78 s, New Download Time Estimate: 0.84 s."
[2023-06-16T09:27:47.970 CEST INFO ][   org.kde.kstars.ekos.capture] - "Received image 2 out of 17."
[2023-06-16T09:27:48.143 CEST INFO ][   org.kde.kstars.ekos.capture] - "Captured /home/ferrante/Light/Luminance/Light_Luminance_30_secs_2023-06-16T09-27-47_005.fits"
[2023-06-16T09:27:48.150 CEST INFO ][   org.kde.kstars.ekos.capture] - Dithering...
[2023-06-16T09:27:48.150 CEST INFO ][   org.kde.kstars.ekos.capture] - "Dithering..."
[2023-06-16T09:27:48.150 CEST INFO ][   org.kde.kstars.ekos.capture] - "Dithering..."
[2023-06-16T09:27:48.151 CEST INFO ][     org.kde.kstars.ekos.guide] - Starting non-guiding dither...
[2023-06-16T09:27:48.151 CEST DEBG ][     org.kde.kstars.ekos.guide] - dither ra_msec: 82 ra_polarity: 1 de_msec: 198 de_polarity: 1
[2023-06-16T09:27:48.151 CEST INFO ][     org.kde.kstars.ekos.guide] - Non-guiding dither successful.
.....
[2023-06-16T09:27:48.451 CEST DEBG ][     org.kde.kstars.ekos.guide] - Executing operation  "Subframing"
[2023-06-16T09:27:48.451 CEST DEBG ][     org.kde.kstars.ekos.guide] - Executing operation  "Capturing"
[2023-06-16T09:27:48.452 CEST DEBG ][     org.kde.kstars.ekos.guide] - Capturing frame...
[2023-06-16T09:27:48.536 CEST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI2600MM Pro : "[DEBUG] Frame ROI x:0 y:0 w:3120 h:2088 "
[2023-06-16T09:27:48.752 CEST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI2600MM Pro : "[DEBUG] Setting frame buffer size to 13029120 bytes. "
**[2023-06-16T09:27:48.753 CEST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI2600MM Pro : "[DEBUG] StartExposure->setexp : 2.000s "**
[2023-06-16T09:27:50.402 CEST DEBG ][   org.kde.kstars.ekos.capture] - Guiding state changed from "Idle" to "Dithering successful"
[2023-06-16T09:27:50.403 CEST INFO ][   org.kde.kstars.ekos.capture] - Dithering succeeded, capture state "Dithering"
[2023-06-16T09:27:50.403 CEST INFO ][   org.kde.kstars.ekos.capture] - "Dithering succeeded."
**[2023-06-16T09:27:50.403 CEST INFO ][   org.kde.kstars.ekos.capture] - "Dithering succeeded."**
[2023-06-16T09:27:50.405 CEST INFO ][   org.kde.kstars.ekos.capture] - "Dither complete."
[2023-06-16T09:27:50.405 CEST INFO ][   org.kde.kstars.ekos.capture] - "Dither complete."
[2023-06-16T09:27:51.155 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capturing 30,000-second Luminance image..."
[2023-06-16T09:27:51.197 CEST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI2600MM Pro : "[DEBUG] Frame ROI x:0 y:0 w:6240 h:4176 "
[2023-06-16T09:27:51.440 CEST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI2600MM Pro : "[DEBUG] Setting frame buffer size to 52116480 bytes. "
[2023-06-16T09:27:51.440 CEST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI2600MM Pro : "[DEBUG] Requested CCD Frame is (0,0) (6248 x 4176) "
[2023-06-16T09:27:51.440 CEST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI2600MM Pro : "[DEBUG] Frame ROI x:0 y:0 w:6248 h:4176 "
[2023-06-16T09:27:51.626 CEST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI2600MM Pro : "[DEBUG] Setting frame buffer size to 52183296 bytes. "
**[2023-06-16T09:27:51.626 CEST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI2600MM Pro : "[DEBUG] Aborting exposure... "**
[2023-06-16T09:27:51.726 CEST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI2600MM Pro : "[DEBUG] StartExposure->setexp : 30.000s "
[2023-06-16T09:27:51.734 CEST INFO ][           org.kde.kstars.indi] - ZWO CCD ASI2600MM Pro :  "[INFO] Taking a 30 seconds frame... 
.....
[2023-06-16T09:28:20.402 CEST INFO ][     org.kde.kstars.ekos.guide] - "Exposure timeout. Restarting exposure..."
[2023-06-16T09:28:20.418 CEST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI2600MM Pro : "[DEBUG] Aborting exposure... "
[2023-06-16T09:28:21.048 CEST DEBG ][           org.kde.kstars.indi] - ZWO CCD ASI2600MM Pro : "[DEBUG] StartExposure->setexp : 2.000s "

Attachments:
[image]
[log_09-24-59.txt](https://github.com/indilib/indi/files/11768017/log_09-24-59.txt)

Version:
KStars 3.6.5 Stable (but also on 3.6.3)
Build: 2023-06-01T18:39:51Z
OS: "ubuntu 22.04"
Kernel Type: "linux"
Kernel Version: "5.15.0-73-generic"
Qt Version: 5.15.3

knro, 2023-06-18:
Thank you for the report but this is not an INDI issue. I will check it out in KStars nevertheless.

closed 2023-06-18 by knro

fenriques, 2023-06-18:
thank you Jasem

fenriques, 2024-03-05:
hi Jasem,
I get this is not an INDI issue, but where I should report Kstars bugs? on KDE bugzilla?
About the issue:
On a recently upgrade Kstars 3.6.9 stable when non-guide dither pulse is enabled the scheduler hangs and the sequence is aborted.
The log simply reports:
[2024-02-25T22:35:31.252 CET INFO ][   org.kde.kstars.ekos.capture] - Dithering...
[2024-02-25T22:35:31.252 CET INFO ][   org.kde.kstars.ekos.capture] - "Dithering..."
[2024-02-25T22:35:31.252 CET INFO ][   org.kde.kstars.ekos.capture] - "Dithering..."
[2024-02-25T22:35:31.252 CET INFO ][     org.kde.kstars.ekos.guide] - Starting non-guiding dither...
[2024-02-25T22:35:31.253 CET WARN ][     org.kde.kstars.ekos.guide] - Non-guiding dither failed.
[2024-02-25T22:35:31.253 CET INFO ][   org.kde.kstars.ekos.capture] - Dithering failed, capture state "Dithering"
[2024-02-25T22:35:31.253 CET INFO ][   org.kde.kstars.ekos.capture] - "Warning: Dithering failed."
[2024-02-25T22:35:31.253 CET INFO ][   org.kde.kstars.ekos.capture] - "Warning: Dithering failed."
[2024-02-25T22:35:47.600 CET INFO ][ org.kde.kstars.ekos.scheduler] - Scheduler is stopping...
[2024-02-25T22:35:47.601 CET INFO ][   org.kde.kstars.ekos.capture] - "CCD capture aborted"

It was first reported on Kstars 3.6.5 (https://github.com/indilib/indi/issues/1908) but in a next release fixed.
It happens again in 3.6.9

Ferrante