indilib/indi #1913: hotplug detecting issue: is it possible to automatically match a device to its INDI driver and device-name by program?

Mas0821, 2023-07-25:
I tried to write a function module to detect if a device is arrived or left, which can remind user whether those necessary devices (CCDs, telescopes, etc.) are connecting well or not while we guiding or keeping exposure to capture the target.
And I designed this module can auto match the driver and device-name when a device is attached (so when it is left, can remind user which device is missing):
- If is a USB device, match device to driver by its vid/pid. Get information from indi-3rdparty/../*.rules.
- If is a serial port device, match device to driver by tty’s checking strings. Get infomations from indi/drivers/telescope, I dig in function which named like Handshack() or checkConnection() to get the “check code (string)”.

But as I dig more, I realized that, it is hard to recognize those devices only by these infomations. 
- For serial port devices (telescopes here), their drivers, some didn’t do Handshack or checkConnection (just “return true” etc.), and some are just check by RA-Format. Which cause one “check code” can match multi drivers & devices (and even find some “wrong drivers” can drives a device successfully, but I don’t know if it works correct, it just moved when I control).
- And for USB devices, some of them, if they came from same manufacturer, they have same vid, and didn’t declare their pid in *.rules. Which means I can’t recognize a device is a CCD or focuser or others if this manufacturer produces multi types of devices. For example, one vid can match a ccd-driver and a focuser-driver, but if I just start both of 2 drivers, I still can’t recognize which USB port is connecting to CCD.

Then I tried figure out how indi get device name while do “start driver ***”, but the trail is dead when I dig into INDI::BaseDevice.
I just speculate, for a device, part of its device-name value is come from the driver I started, cus it is different when I start different drivers (some device can use multi drivers to drive), and the part of device-name always same is finally get from “USB Device descriptor”, from some key’s value, like the value of “iProduct”.

I think I might miss something important, really need some help or hint.
ps: Sorry for my bad English (not my first language), if there has anything I described unclear, free to ask.
pss: Notice that latest indi added attach() and detach() in BaseDevice, maybe indi will support hotplug detect in future?

knro, 2023-07-25:
Thank you for the report. This is a known issue. If you have USB device with VID:PID then you know which device this, and which driver to use for it. But for serial devices that just use some generic Prolific serial to USB chip, it is near impossible for several reasons:
+ Most chips are generic and have no relationship at all to what they are controlling.
+ A possible trick if you want to automatically detect is to run Handshake for all possible drivers and not only this will take lots of time, but some devices would completely hang if they receive unknown commands and it would not be possible to communication with them again unless power is cycled.

Another method used by INDI Web Manager Serial Assistant tool is to ask the users to plug devices one by one then we will be able to detect that this serial port belongs to this device and so worth.

Mas0821, 2023-07-27:
Thanks for ur reply !!!
That is much the same as my perception,  I'll put this function module on ice until I find other ways to achieve similar effects.

closed 2023-07-27 by Mas0821