indilib/indi #1919: Sesto Senso 2 - cannot process position info

Planet30, 2023-08-09:
Sesto Senso 2 connected with USB RPi4B running EKOS.

I keep receiving this warning in the log file
... Sesto Senso 2 @ 192.168.68.55: [WARNING] Failed to process position response: _STEP80933 (10 bytes)

In EKOS, the absolute position line/entry has a red dot (virtual LED), the value is 0.

[image]

knro, 2023-08-10:
Update its firmware and update INDI to v2.0.3

closed 2023-08-10 by knro

Planet30, 2023-08-10:
FW has been up to date.

INDI is running on the built-in RPi of the StarGO2 controller of the mount.  I am not supposed to update it.  I have to check with Avalon.

Planet30, 2023-08-10:
Is there way to just update the driver without the rest of INDI?

Planet30, 2023-08-10:
There are actually plenty of other error messages ... not just the warning shown above.

knro, 2023-08-10:
Not sure how, I think it's best to reach out to Avalon to resolve this.

Planet30, 2023-08-10:
OK... stupid question... (not really an expert on RPi's). How do I just update INDI to version 2.0.x and its latest drivers without updating the whole OS?

I could only find guides that either _install_ INDI, _update_ the OS or _upgrade_ the whole systems.

Planet30, 2023-08-10:
It seems that I cannot update my systems to INDI version 2.0.3... it keeps telling me I have the latest version, which is 1.9.7. ...?!

**sudo apt-get install -y indi-full gsc**
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gsc is already the newest version (1.3+deb10u1).
indi-full is already the newest version (1.9.7+deb10u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Brusin64, 2024-11-13:
Ciao,
I had the same problem with Sesto senso 2, I solved upgrading indi from 1.9.7 to 2.1.0.
I followed this thread:
https://stargazerslounge.com/topic/423938-guide-installing-indi-and-kstarsekos-on-raspberry-pi-5-the-proper-way/
in particular these steps:

git clone --depth 1 https://github.com/indilib/indi.git

cd indi

cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug indi

make -j4

sudo make install

Check indi version before and after the upgrade with following command:

indiserver --version

Good luck.