indilib/indi #1921: INDI build on Windows

AstroAir, 2023-08-12:
Recently, due to writing an astronomical software, I have already written the client for INDI, but when I tried to write it on Windows, the construction of INDI was completely unsuccessful. Because at the time of design, it seems that INDI did not consider its use on Windows at all. I compiled using Mingw64 (Msys2), but still cannot pass. Most of the problems are focused on using network communication libraries unique to Linux, such as `sys/socket`. h. Currently, I have started migrating under Windows, but under Mingw64, there is a method of building

### Install Msys2
Download https://github.com/msys2/msys2-installer/releases/download/2023-07-18/msys2-x86_64-20230718.exe

```shell
pacman -Syu # Update all of the packages
pacman -S mingw-w64-x86_64-toolchain # Install g++ toolchain
pacman -S mingw-w64-x86_64-cmake # Install Cmake
```

### Install dependencies

If there is anything missing, you can use a search engine to search for the name of the Msys2+package (usually between lib and - dev)

```shell
pacman -S mingw-w64-x86_64-zlib
pacman -S mingw-w64-x86_64-cfitsio
pacman -S mingw-w64-x86_64-libnova
pacman -S mingw-w64-x86_64-libusb
pacman -S mingw-w64-x86_64-fftw
pacman -S mingw-w64-x86_64-libxisf
pacman -S mingw-w64-x86_64-rtl-sdr
```

#### Install libev-dev

This is what Indiserver relies on, and other components do not seem to need it

Download sources from http://dist.schmorp.de/libev/Attic/

```shell
pacman -S autoconf
pacman -S automake
pacman -S libtool
pacman -S make
sh autogen.sh
./configure
make
make -j4
make install
```

### Build the project

After installing all of the dependencies , we can try to build the sources manually

```shell
cmake ..
cmake --build . -jn # or make -jn
```

But by default, only `INDIClient` will be built, so we need to modify `CMakeLists.txt` first

```
if(INDI_BUILD_DRIVERS)
    if(WIN32 OR ANDROID) # Remove WIN32 then allow the cmake to build the whole project
        message(WARNING "INDI drivers are only supported under Linux, BSD, MacOS, and Cygwin while current system is " ${CMAKE_SYSTEM_NAME})
    else(WIN32 OR ANDROID)
    ...
endif()
```

then retry building commands.

like tools/setINDIProperty.cpp can be written as following

```cpp
/* connect to an INDI server and set one or more device.property.element.
 */
[... 626 lines cut ...]
            usage();
        }

        sp->ev            = (SetEV *)realloc(sp->ev, (sp->nev + 1) * sizeof(SetEV));
        sp->ev[sp->nev].e = strdup(ev);
        sp->ev[sp->nev].v = strdup(e);
        if (verbose > 1)
            fprintf(stderr, "Found assignment %s=%s\n", sp->ev[sp->nev].e, sp->ev[sp->nev].v);
        sp->nev++;

        ev = s;

    } while (!last);
}

/* send each SetSpec, all of which have a known type, to wfp
 */
static void sendSpecs(FILE *wfp)
{
    int i;

    for (i = 0; i < nsets; i++)
        sendNew(wfp, sets[i].dp, &sets[i]);
}
```

knro, 2023-08-12:
It would be great if INDI can be built on Windows (not just INDI Client like it is now). Just to be clear, INDI Client compiles and runs on Windows just fine without any additional libraries. It is that INDI server + drivers that do not compile natively on Windows. I presume you are trying to tackle the INDI server + drivers part.

AstroAir, 2023-08-12:
@knro Yes, there are no issues with the construction of INDIClient. At present, I am working on other parts of the indiserver, but due to some confusion in the code, it has not been resolved yet. The current main issue is still focused on `indidriver`, and the previous ones can already be built normally

knro, 2023-08-12:
Alright, please let us know if you can make INDI runs natively on Windows, this would be quite a breakthrough. We still have the issues of dependencies but once INDI runs natively, the rest of the issues can be streamlined in the build process.

closed 2023-08-15 by AstroAir

laheller, 2023-10-05:
@knro @AstroAir 

With only minimal chages in the sources I successfully built INDI on Windows/[Cygwin](https://cygwin.com/). More info and PR later after testing.

[image]

BR,

Ladislav

knro, 2023-10-05:
Very excited to see this @laheller ! Were you able to test any drivers with any clients?

laheller, 2023-10-05:
@knro 

Not yet, because I am fighting with the indiweb installation in python/cygwin environment and it has a troubles with psutil dependency.
Once I have test results, will let you know.
First probably here in this issue thread.

BR,

Ladislav

knro, 2023-10-06:
You can still test everything *without* indiweb.

laheller, 2023-10-06:
@knro 

Finally I managed to work also indiweb on Cygwin beside one issue (see the screenshot).
Tested with
- KSTARS/EKOS client on Windows
- INDI server running on Windows/Cygwin
- Simulators added to the profile

Everything seems to be working.

Screenshot - indiweb started server and simulator devices:
[image]

Additional screenshots from EKOS:
[image]

[image]

What do you think, do I need to perform any further tests?

BR,

Ladislav

laheller, 2023-10-06:
> Wow that's very nice! Have you tried connecting to any physical devices via Serial connection or USB??

Not yet but I definitely will try with my:
- SkyWatcher HEQ5 mount
- Canon DSLR camera
- ZWO ASI camera

Probably not under night sky but just "blindly" instead. Once done, will let you know.

knro, 2023-10-06:
I think this might work just for the skywatcher. The Canon needs libgphoto2 and ZWO driver does not include any Windows DLLs so it can work on Windows. Perhaps you can try to add Windows support to this driver.

laheller, 2023-10-07:
@knro 

While I am checking the devices and their drivers, here is the [PR](https://github.com/indilib/indi/pull/1950). I just did it, because it is only for this repo and also independent from the 3rd party drivers.

BR,

Ladislav

laheller, 2023-10-09:
Hi @knro 

I was able to build also **indi-asi** drivers on Cygwin.
I have a **ZWO ASI 120 MC-S** camera without cooling.
When I start EKOS and open Indi Control Panel, I see an error message there which says:

> Failed to get perc power information (ASI_ERROR_INVALID_CONTROL_TYPE).

Also in the Indi Control Panel / Image Info tab I see that main camera parameters are incorrectly detected, because the maximum width is definitely not 2. See the screenshot below.
The correct values should be:

> Maximum width: 1280
> Maximum height: 960
> Pixel size X/Y: 3.75 μm
> 

[image]

Please if you have time, check & answer also my previous comment above regarding the Kstars/Windows crash.
If you are not familiar with Windows, please redirect me to the developer, who is maintaining the Kstars/Windows builds, because I have an issue with Kstars and DSLR camera which I cannot workaround.

Best Regards,

Ladislav