indilib/indi #1923: Crash on indi connection

FalCT60, 2023-08-20:
**Describe the bug**
When clicking the connect to indi button from ekos, the connection fails.
This suddenly occured after last update.

**Expected behavior**
Classic connection to indi server and access to the capture etc. tabs.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Stellarmate 64 bits for PI
 - Version 1.7.8 downloaded 2023 aug 10th

**Log Files**
[indi_app_log_23-08-20T09-49-09.txt](https://github.com/indilib/indi/files/12387805/indi_app_log_23-08-20T09-49-09.txt)
[indi_debug_log_23-08-20T09-49-09.txt](https://github.com/indilib/indi/files/12387806/indi_debug_log_23-08-20T09-49-09.txt)
[kstars_debug_log_2023-08-20T09-48-21.txt](https://github.com/indilib/indi/files/12387807/kstars_debug_log_2023-08-20T09-48-21.txt)
[log_09-48-21.txt](https://github.com/indilib/indi/files/12387808/log_09-48-21.txt)
[log_09-49-59.txt](https://github.com/indilib/indi/files/12387809/log_09-49-59.txt)
[log_09-54-17.txt](https://github.com/indilib/indi/files/12387810/log_09-54-17.txt)
[log_10-32-24.txt](https://github.com/indilib/indi/files/12387818/log_10-32-24.txt)
[log_10-34-57.txt](https://github.com/indilib/indi/files/12387819/log_10-34-57.txt)

knro, 2023-08-20:
From StellarMate App, go to Device --> Purge Driver Config

or try to delete this file ~/.indi/ZWO CCD.xml

does this help?

FalCT60, 2023-08-20:
I restored the image straight after my report, tested that it worked flawlessly and then started the update.
It is not finnished yet, about 20% left so far, so I cannot test for the moment.
However, among my many actions I did remove everything in .ZWO, .indi and some parts in .local/.../kstars.
The first start next that was successful, but then the next ones failed.
I am waiting for the update to end, then I test and come back.

FalCT60, 2023-08-20:
Update finished a few minutes ago.
Tested as is with the only optical train created before : worked fine 4~5 times
Then created all remaining optical trains : failed.

knro, 2023-08-20:
Ok submit a ticket with StellarMate

FalCT60, 2023-08-20:
I do not understand there... sorry.
Here is the content of .indi (.txt to let it pass through)
[profiles.db.txt](https://github.com/indilib/indi/files/12388233/profiles.db.txt)
[ZWO CCD ASI183MC_config.xml.txt](https://github.com/indilib/indi/files/12388234/ZWO.CCD.ASI183MC_config.xml.txt)
[ZWO CCD ASI183MC_config.xml.default.txt](https://github.com/indilib/indi/files/12388235/ZWO.CCD.ASI183MC_config.xml.default.txt)
and the dtatabase into .local/../kstars
[userdb.sqlite.txt](https://github.com/indilib/indi/files/12388237/userdb.sqlite.txt)
And... clearing the conents of ~/.indi/ makes it work the first ime after, then again fails.

FalCT60, 2023-08-22:
Fixed since yesterday.
Thanks for your awesome work.

closed 2023-08-22 by FalCT60