indilib/indi #1925: INDI Control Panel doesn't persistently save camera gain and offset from session to session

Ruegs, 2023-08-21:
**Describe the bug**
I am using an ASI6200MM, and KStars/EKOS 3.6.6 and 3.6.7 for image capture. If I go into the INDI Control Panel and set and save the camera gain, that value follows through to the EKOS capture module. But when I close and restart KStars/EKOS, I get one of four different gain values. First, the one I set in the last session via the INDI Control Panel (which I believe is what it should always be). Second, sometimes I get the gain I use for polar alignment. Third, sometimes I get the gain I use for focusing. Finally, I get gain 100 (which I don't generally use).

**To Reproduce**
1) In the INDI control panel I set gain = 0 and then save the configuration.
2) I disconnect the devices in EKOS (not physical disconnection) and stop EKOS.
3) I restart EKOS and go to the INDI control panel and the gain is still 0 - all good.
4) I go to the capture module and enter a simple one frame sequence with gain = 100. I DO NOT change the settings I had SAVED in the INDI control panel (i.e., gain = 0).
5) I again disconnect the devices in EKOS and stop EKOS.
6) I restart EKOS and go to the INDI control panel and the gain is now 100 - I think this is a bug. In my mind INDI should retain the saved configuration (gain = 0 in this example) until I change it. In my case, since I use different gain values in the EKOS Alignment Module and in the EKOS Focus Module, it seems that the INDI control panel just picks up whatever the last gain value was and stores that in the INDI control panel - I don't even save those values but there they are the next time I start up.

**Expected behavior**
The camera gain and offset values set in the INDI control panel should persist until changed by the user intentionally. If a user uses a different gain or offset value to shoot a sequence of frames (or for alignment or focusing) those values should not reset the user saved values in the INDI control panel

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: MacOS 13.4.1 (c)
 - Version: INDI 2.0.3, KStars/EKOS 3.6.6 and 3.6.7

**Log Files**
N/A

knro, 2023-08-22:
Thank you for the report. I don't think this is an INDI issue. It's probably Ekos trying to set all the different gain per each module. INDI by itself is working as intended.

closed 2023-08-22 by knro

Ruegs, 2023-08-22:
Thanks Jasem,

I had posted this over on the forum as well - is something else I should do with this. I think the behavior is confusing and it would be great to have it addressed.

BTW - thanks for all you do!

Fred