indilib/indi #1928: Onstep controller Indi issue

1crem1, 2023-08-27:
Hello
I am currently trying to use an off the shelf Onstep controller with Kstars. I am using the LX200 OnStep driver. I can connect however once initialisation occurs I receive a repeated warning on the Indi driver control panel (in orange):

Date/time: [WARNING] Communication: Ft# error, check connection.
Date/time: [WARNING] Invalid response, check connection.

This error message repeats every second in the Indi driver control panel, however the mount appears to respond to commands using the mount control feature in the mount section of Ekos.
I haven’t yet used it in the field.

I’m not sure how to progress. The ‘connection’ tab reveals nothing that I can detect.

Thanks
Andrew

azwing, 2023-08-29:
Hi,

can se continue this discussion on the OnStep Forum
https://www.indilib.org/forum/development/1406-driver-onstep-lx200-like-for-indi.html
It will be easier to manage.
At least would be helpfull to have:
- Onstep Firmware Version
- Indi logs 
- at a first glance the Ft# error could be related to Focuser.

azwing, 2023-09-11:
After analysis of the issue it is clearly a firmware issue and not a driver bug.
The error messages clearly are triggered when the firmware doesn't answer to the request (in this case the Focuser Temperature).
Normal firmware behavior is to reply either with the real value or by default with 10.0# so that no error messages are issue by the driver.

I consider this issue as closed

closed 2023-09-12 by knro