indilib/indi #1932: Rainbow Astro Simulate Pier Side not changing Pier Side on Meridian Flip

gordoabc, 2023-09-08:
**Describe the bug**
When Simulate Pier Side is selected in the driver control panel it correctly indicates current setting of pier side.  When  Meridian Flip is executed the Pier See is not changed.  This causes the mount to stop moving if an HA limit is set and Guiding doesn't work once the flip has happened

**To Reproduce**
Select "Simulate Pier Side" in the Rainbow Astro telescope driver control panel
Execute a goto to the other side of the pier
check pier side variable in the telescope control panel

**Expected behavior**
The Pier Side variable should change when the pier side changes

**Desktop (please complete the following information):**
 - OS: Ubuntu LTS 22.05
 - Version kstars/ekos 3.6.6

**Log Files**
[log_18-59-34.txt](https://github.com/indilib/indi/files/12559945/log_18-59-34.txt)

knro, 2023-09-10:
I contacted Rainbow to ask them about this. Maybe they can natively report pier side now and no need for simulation? Let's see. If you can ask them the same question that would be great.

gordoabc, 2023-09-10:
Happy to ask - do you have an email to contact them?

gordoabc, 2023-09-12:
> Ok I pushed preliminary implementation for this but it's completely untested. Please try it out.

Will do - traveling so it will be this weekend before I can try anything - testing may be limited by clouds but should at least be able to confirm pier side changes.Thanks

knro, 2023-09-13:
You can test indoors

gordoabc, 2023-09-15:
OK - must be doing something stupid - when I switched to the nightly repository it doesn't pull the new version:

thomas@mele:~$ sudo apt-get upgrade indi-full
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
indi-full is already the newest version (2.0.3+202308010715~ubuntu22.04.1).
Calculating upgrade... Done
The following packages have been kept back:
  gjs libgjs0g openssh-client openssh-server openssh-sftp-server thermald
The following packages will be upgraded:
  fxload kstars-bleeding kstars-bleeding-data kstars-bleeding-dbg
  libomegonprocam libstellarsolver
6 upgraded, 0 newly installed, 0 to remove and 6 not upgraded.
Need to get 211 MB of archives.
After this operation, 2,802 kB of additional disk space will be used.

Do I have to compile the driver instead?

gordoabc, 2023-09-15:
I compiled to driver and this got me to version 1.3 of the driver.  IT does not seem to be getting a Pier Side:

[2023-09-14T20:48:32.825 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:CG3-000.8580> "
[2023-09-14T20:48:32.826 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] CMD <:CY#> "
[2023-09-14T20:48:32.841 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:CY+141.14/-061.96> "
[2023-09-14T20:48:33.092 MDT DEBG ][     org.kde.kstars.ekos.mount] - Ha:  -1.86925  haLimit  0.3   "Pier Side: Unknown"  haLimitReached  false  lastHa  -1.88406
[2023-09-14T20:48:33.092 MDT DEBG ][     org.kde.kstars.ekos.mount] - Mount status changed from  "Slewing"  to  "Tracking"
[2023-09-14T20:48:33.095 MDT DEBG ][     org.kde.kstars.ekos.mount] - Slew finished, MFStatus  "MOUNT_FLIP_NONE"
[2023-09-14T20:48:33.841 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] CMD <:GR#> "
[2023-09-14T20:48:33.847 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:GR21:12:12.7> "
[2023-09-14T20:48:33.847 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] CMD <:GD#> "
[2023-09-14T20:48:33.863 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:GD+37*59'05.9> "
[2023-09-14T20:48:33.863 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] CMD <:GZ#> "
[2023-09-14T20:48:33.879 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:GZ075*44'52.9> "
[2023-09-14T20:48:33.879 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] CMD <:GA#> "
[2023-09-14T20:48:33.896 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:GA+67*29'10.5> "
[2023-09-14T20:48:33.897 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] CMD <:CG3#> "
[2023-09-14T20:48:33.911 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:CG3-000.8580> "
[2023-09-14T20:48:33.911 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] CMD <:CY#> "
[2023-09-14T20:48:33.930 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:CY+141.14/-061.97> "
[2023-09-14T20:48:34.093 MDT DEBG ][     org.kde.kstars.ekos.mount] - Ha:  -1.86881  haLimit  0.3   "Pier Side: Unknown"  haLimitReached  false  lastHa  -1.86925
[2023-09-14T20:48:34.927 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] CMD <:GR#> "
[2023-09-14T20:48:34.934 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:GR21:12:12.7> "
[2023-09-14T20:48:34.934 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] CMD <:GD#> "
[2023-09-14T20:48:34.949 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:GD+37*59'05.9> "
[2023-09-14T20:48:34.950 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] CMD <:GZ#> "
[2023-09-14T20:48:34.965 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:GZ075*44'54.6> "
[2023-09-14T20:48:34.966 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] CMD <:GA#> "
[2023-09-14T20:48:34.982 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:GA+67*29'23.2> "
[2023-09-14T20:48:34.982 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] CMD <:CG3#> "
[2023-09-14T20:48:34.998 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:CG3-000.8580> "
[2023-09-14T20:48:34.999 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] CMD <:CY#> "
[2023-09-14T20:48:35.013 MDT DEBG ][           org.kde.kstars.indi] - Rainbow RST-135 : "[DEBUG] RES <:CY+141.14/-061.97> "
[2023-09-14T20:48:35.092 MDT DEBG ][     org.kde.kstars.ekos.mount] - Ha:  -1.86853  haLimit  0.3   "Pier Side: Unknown"  haLimitReached  false 

I homed it and drove around a bit.  Tried meridian flip calibrate in PHD2 but it also flags Pier Side Unknown.

gordoabc, 2023-09-15:
[log_18-47-13.txt](https://github.com/indilib/indi/files/12615085/log_18-47-13.txt)

knro, 2023-09-15:
Please try to compile again and report back

closed 2023-09-15 by knro

gordoabc, 2023-09-15:
Still reports Pier Side unknown in log and PHD2 meridian flip calibrate
[log_05-32-49.txt](https://github.com/indilib/indi/files/12618897/log_05-32-49.txt)

commit f821d7fb (2023-09-15): Fix response parse. Fixes #1932