indilib/indi #1933: Falcon Rotator only de-rotates in one direction

lpldave, 2023-09-10:
When using the derotation feature for the Pegasus Falcon Rotator, it only moves in the counterclockwise direction regardless of whether the direction is set to reversed or not.  

To Reproduce:

Launch the Falcon Rotator driver through the device manager in KStars.  Set value for derotation (I used 10 ms per step).  It rotates in the counterclockwise direction.  Set value for derotation back to zero to disable it.  Click "Enable" under "Reverse" and set value for derotation (I used 10 ms per step).  Rotation is still in the same direction, but is expected to be reversed.

Desktop:

 - OS: PopOS 22.04 LTS (based on Ubuntu)
 - INDILib 2.0.3 and KStars 

Log File:

INFO	7.563281 sec	: Session log file ***
INFO	12.550966 sec	: Pegasus Falcon is online.
INFO	478.595166 sec	: De-rotation is enabled and set to 1 step per 10 milliseconds.
INFO	490.119361 sec	: De-rotaiton is disabled.
INFO	493.575854 sec	: Rotator direction is reversed.
INFO	498.519098 sec	: De-rotation is enabled and set to 1 step per 10 milliseconds.
INFO	523.759419 sec	: De-rotaiton is disabled.
INFO	525.263378 sec	: Rotator direction is normal.
INFO	530.227448 sec	: De-rotation is enabled and set to 1 step per 10 milliseconds.
INFO	555.030551 sec	: De-rotaiton is disabled.
INFO	568.352810 sec	: Pegasus Falcon is offline.

escribana, 2023-09-22:
Hi @lpldave

I just checked the reaction of my Falcon Rotator respective to the reverse direction and did not notice any problems. Following your suggestion I set the derotation to 10ms per step and changed the rotator direction according your proceeding. The device is reacting as expected, rotates backward if the direction is reversed and forward if set to normal.

OS: Official Ubuntu 22.04 LTS
INDILib 2.0.3 and KStars 3.6.7beta

lpldave, 2023-09-22:
Thank you for checking!  I’ll give mine another try and see if I can get any more diagnostic info about what might be going on with my setup.

escribana, 2023-09-23:
Please set the following in "Logging" (button [Logs] in EKOS):

1. Verbosity: Verbose
2. Output: File
3. Drivers: Rotator

Open [Pegasus Falcon]/[Options] in "INDI Control Panel" and set the following:

1. Debug Levels: All (i.e. add "Driver Debug")
2. Logging Levels:  All (i.e. add "Driver Debug")
3. Log Output: "To Log File"

Now the emitted low level commands will be saved to something like this:
[INFO] Session log file /home/<user>/.indi/logs/2023-09-23/indi_falcon_rotator/indi_falcon_rotator_07:38:42.log

lpldave, 2023-10-27:
Hi, sorry for the huge delay on this.  I tried it again with logging, and the file is attached below.  I tried both reversing in the middle of rotation, as well as stopping rotation, reversing, and restarting.  In all cases, it still only moved in the counterclockwise direction.

INFO	10.833675 sec	: Session log file /home/obrien/.indi/logs/2023-10-27/indi_falcon_rotator/indi_falcon_rotator_08:11:57.log
DEBUG	11.037204 sec	: CMD <FA>
DEBUG	11.076262 sec	: RES <FR_OK:7280:84.03:0:1:0:0
>
DEBUG	12.081354 sec	: CMD <FA>
DEBUG	12.116055 sec	: RES <FR_OK:7280:84.03:0:1:0:0
>
DEBUG	13.121240 sec	: CMD <FA>
DEBUG	13.156261 sec	: RES <FR_OK:7280:84.03:0:1:0:0
>
DEBUG	14.161204 sec	: CMD <FA>
DEBUG	14.197016 sec	: RES <FR_OK:7280:84.03:0:1:0:0
>
DEBUG	15.201199 sec	: CMD <FA>
DEBUG	15.236746 sec	: RES <FR_OK:7280:84.03:0:1:0:0
>
DEBUG	16.241196 sec	: CMD <FA>
DEBUG	16.276979 sec	: RES <FR_OK:7280:84.03:0:1:0:0
>
DEBUG	17.281193 sec	: CMD <FA>
DEBUG	17.317570 sec	: RES <FR_OK:7280:84.03:0:1:0:0
>
DEBUG	18.322811 sec	: CMD <FA>
DEBUG	18.357489 sec	: RES <FR_OK:7280:84.03:0:1:0:0
>
DEBUG	18.805417 sec	: CMD <DR:10>
INFO	18.805485 sec	: De-rotation is enabled and set to 1 step per 10 milliseconds.
DEBUG	19.360483 sec	: CMD <FA>
DEBUG	19.397779 sec	: RES <FR_OK:7304:84.30:1:1:1:0
>
DEBUG	20.401200 sec	: CMD <FA>
DEBUG	20.437990 sec	: RES <FR_OK:7349:84.82:0:1:1:0
>
DEBUG	21.439211 sec	: CMD <FA>
DEBUG	21.478221 sec	: RES <FR_OK:7394:85.34:1:1:1:0
>
DEBUG	22.479639 sec	: CMD <FA>
DEBUG	22.518768 sec	: RES <FR_OK:7439:85.86:1:1:1:0
>
DEBUG	23.520536 sec	: CMD <FA>
DEBUG	23.558740 sec	: RES <FR_OK:7485:86.39:0:1:1:0
>
DEBUG	24.560486 sec	: CMD <FA>
DEBUG	24.599423 sec	: RES <FR_OK:7530:86.91:1:1:1:0
>
DEBUG	25.600699 sec	: CMD <FA>
DEBUG	25.639758 sec	: RES <FR_OK:7575:87.43:1:1:1:0
>
DEBUG	26.641495 sec	: CMD <FA>
DEBUG	26.680008 sec	: RES <FR_OK:7621:87.96:0:1:1:0
>
DEBUG	27.681144 sec	: CMD <FA>
DEBUG	27.720259 sec	: RES <FR_OK:7666:88.48:1:1:1:0
>
DEBUG	28.721686 sec	: CMD <FA>
DEBUG	28.760582 sec	: RES <FR_OK:7711:89.00:1:1:1:0
>
DEBUG	29.762060 sec	: CMD <FA>
DEBUG	29.800478 sec	: RES <FR_OK:7756:89.52:1:1:1:0
>
DEBUG	30.801782 sec	: CMD <FA>
DEBUG	30.841061 sec	: RES <FR_OK:7802:90.05:0:1:1:0
>
DEBUG	31.842844 sec	: CMD <FA>
DEBUG	31.880825 sec	: RES <FR_OK:7847:90.57:1:1:1:0
>
DEBUG	32.882365 sec	: CMD <FA>
DEBUG	32.921553 sec	: RES <FR_OK:7892:91.09:1:1:1:0
[... 190 lines cut ...]
DEBUG	93.215242 sec	: CMD <FA>
DEBUG	93.255142 sec	: RES <FR_OK:7077:81.68:0:0:0:1
>
DEBUG	94.256329 sec	: CMD <FA>
DEBUG	94.295406 sec	: RES <FR_OK:7077:81.68:0:0:0:1
>
DEBUG	94.701430 sec	: CMD <FN:0>
DEBUG	94.727479 sec	: RES <FN:0
>
INFO	94.727521 sec	: Rotator direction is normal.
DEBUG	95.296147 sec	: CMD <FA>
DEBUG	95.335594 sec	: RES <FR_OK:7077:81.68:0:0:0:0
>
DEBUG	96.336765 sec	: CMD <FA>
DEBUG	96.375870 sec	: RES <FR_OK:7077:81.68:0:0:0:0
>
DEBUG	97.377005 sec	: CMD <FA>
DEBUG	97.416152 sec	: RES <FR_OK:7077:81.68:0:0:0:0
>
DEBUG	98.417617 sec	: CMD <FA>
DEBUG	98.456455 sec	: RES <FR_OK:7077:81.68:0:0:0:0
>
DEBUG	99.457207 sec	: CMD <FA>
DEBUG	99.496654 sec	: RES <FR_OK:7077:81.68:0:0:0:0
>

escribana, 2023-10-29:
Hi @lpldave

The analysis of the log file shows that all is correct!
Look here:  [RS-LogAnalysis.txt](https://github.com/indilib/indi/files/13197597/RS-LogAnalysis.txt)
Thus I don't know why you experience a wrong behaviour while the device is reporting a correct one.  I'm out of ideas!

lpldave, 2023-10-31:
Thanks for checking.  I haven’t seen that particular error report you linked, but I have seen a few others.  Looks like it’s basically the same issue I’m having.  I’ll try testing this on a Windows machine when I get a chance, and send a message to tech support.

lpldave, 2023-11-18:
I was able to get the Windows driver running in a virtual machine under PopOS/Ubuntu.  The derotation worked in both directions, so it doesn't seem to be a hardware problem.

escribana, 2023-11-18:
Ok, so it seems to be a publication problem!!! (-> [Forum](https://pegasusastro.com/forum/viewtopic.php?t=91).
The published  [Falcon_Serial_Command_Table.pdf](https://github.com/indilib/indi/files/13400359/Falcon_Serial_Command_Table.pdf) seems not to be complete. I don't know why they want to hide certain commands.:-(
The only way to reveal these would be a very time consuming reverse engeneering.

closed 2024-04-05 by github-actions[bot]

lpldave, 2024-07-16:
I figured out what the problem was.  The driver only takes positive values of the de-rotation parameter (milliseconds per step), which drives the rotator in the counterclockwise direction.  Negative values are required to move it in the clockwise direction.  I confirmed this by directly sending commands to it over the USB port.  A command of "DR:1" as I used above gives 1 ms/step rotation in the counterclockwise direction.   A command of "DR:-1" gives 1 ms/step rotation in the clockwise direction.

So the driver should be modified to allow for both positive and negative values of the de-rotation parameter.

escribana, 2024-07-17:
Many thanks of the analysis! So the published command table is at least "imprecise". I will upload a MR for INDI driver V1 (newest rotator is V2) and hopefully all is well afterwards.