indilib/indi #1938: 10micron mount doesn't stop tracking when commanded

gordoabc, 2023-09-24:
When my 10micron GM1000HPS is tracking after a slew if I turn off tracking in the ekos mount panel or in the Indi control panel it gives an error:

[WARNING] Tracking state is not supported. 

and continues tracking.

**To Reproduce**
Exact steps to reproduce the behavior.
1. in kstars perform a goto
2. once it has slewed and is tracking select the OFF button under tracking in the LX200 10 Micron main control panel

**Expected behavior**
It stops tracking when OFF is pressed

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 LTS
 - Version 3.6.6 (indilib 2.0.3 10micron driver version 1.1)
 - 10micron firmware 3.1.10

**Log Files**
[log_16-16-03.txt](https://github.com/indilib/indi/files/12709816/log_16-16-03.txt)

d33psky, 2023-09-30:
Thanks for reporting.
Found the issue; it's an omission in the 10micron driver with a fallthrough to the lx200 driver that shows just the warning message.
Should be an easy fix.

d33psky, 2023-09-30:
PR https://github.com/indilib/indi/pull/1945

d33psky, 2023-09-30:
time flows from bottom to top
```
2023-09-30T18:50:55: [INFO] Gstat changed from 7 to 0 
2023-09-30T18:50:54: [INFO] Start tracking. 
2023-09-30T18:50:48: [INFO] Gstat changed from 0 to 7 
2023-09-30T18:50:48: [INFO] Stop tracking. 
```

closed 2023-10-05 by d33psky