indilib/indi #1947: Can't save configuration because indipropertybasic's & overload is protected

ktgilliam, 2023-10-05:
**Describe the bug**
Can't save a property to the configuration file in saveConfigItems(), as shown [here](http://docs.indilib.org/drivers/properties.html):

```
bool MyCustomDriver::saveConfigItems(FILE *fp)
{
    INDI::DefaultDevice::saveConfigItems(fp);
    IUSaveConfigText(fp, &WhatToSayTP);

    return true;
}
```

because the & overload is protected. Compiling as shown generates the error:

`error: ‘INDI::PropertyView<T>* INDI::PropertyBasic<T>::operator&() [with T = _IText]’ is protected within this context
[build]   745 |     IUSaveConfigText(fp, &WhatToSayTP);`

Removing the ampersand gives a deprecation warning:
`warning: ‘INDI::Property::operator INDI::PropertyViewText*() const’ is deprecated: Do not use INDI::Property as pointer. [-Wdeprecated-declarations]
`

**Expected behavior**
Example code should compile without errors or warnings.

pawel-soja, 2023-10-06:
It is highly probable that `WhatToSayTP` is of a new type, hence the error.

@ktgilliam Instead
```cpp
IUSaveConfigText(fp, &WhatToSayTP);
```
use
```cpp
WhatToSayTP.save(fp);
```

ktgilliam, 2023-10-10:
Got it - that did it, thanks.

Should the tutorial at docs.indilib.org be updated, and if so, is this git issue the correct place to put a ticket for that? If not, I will go ahead and close it.

Thanks again!

pawel-soja, 2023-10-10:
@ktgilliam oh, I see. Thank you! I will make a fix in the documentation soon.

closed 2023-10-10 by ktgilliam