indilib/indi #1949: Following the script in the readme to set up my Raspberry to try and start development gives error : Note, selecting 'libgsl-dev' instead of 'libgsl0-dev' E: Unable to locate package libdc1394-dev

ian-morgan99, 2023-10-06:
I am following the instructions in the readme to start helping to make changes to give support to more devices.

When I follow the script:

sudo apt-get install -y \
   git \
   cdbs \
   dkms \
   cmake \
   fxload \
   libev-dev \
   libgps-dev \
   libgsl-dev \
   libraw-dev \
   libusb-dev \
   zlib1g-dev \
   libftdi-dev \
   libgsl0-dev \
   libjpeg-dev \
   libkrb5-dev \
   libnova-dev \
   libtiff-dev \
   libfftw3-dev \
   librtlsdr-dev \
   libcfitsio-dev \
   libgphoto2-dev \
   build-essential \
   libusb-1.0-0-dev \
   libdc1394-dev \
   libboost-regex-dev \
   libcurl4-gnutls-dev \
   libtheora-dev

I get the error:

Note, selecting 'libgsl-dev' instead of 'libgsl0-dev'
E: Unable to locate package libdc1394-dev

From the research I've done so far, it seems that package gets renamed from time to time. Does the documentation need an update?
Interestingly, I get the same error if I try to install a partial list of these packages also up to the preceding line.

This is on a Raspberry Pi 4b 8gb

knro, 2023-10-10:
libdc1394-dev should exist fine on Bullseye

closed 2023-10-10 by knro

ian-morgan99, 2023-10-11:
Unfortunately, the Astroberry server build that is currently released, is on buster.

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Is there a trick to installing libdc1394-dev under buster?

commit 63757a06 (2023-10-10): Remove old package name. Fixes #1949