indilib/indi #1951: Esatto focuser driver temperature does not update.

gordoabc, 2023-10-10:
**Describe the bug**
I added a temperature sensor to my Esatto 3 focuser so I can implement adaptive focus. The driver reads a sensible looking temperature when it loads but does not update. If I quit kstars and then reload it gets a new temperature that tracks the known temperature so I think it is reading OK, it just doesn’t change from the initial temp when the driver starts.

The motor temperature does update.

Another user reports:  "I have noticed the same behaviour with 2 different focusers (an Esatto2 and an Esatto3), with of course 2 different temperature sensors. Not only I see the defect with Kstars, but also with CCDCiel . The motor temperature is updated on a regular basis, while the outside temperature is stuck to its initial value."

https://www.indilib.org/forum/focusers-filter-wheels/13932-esatto-focuser-temperature-not-updating.html

**To Reproduce**
Run Esatto driver
note temperature in driver panel or focuser panel
wait for outside temperature to change and note it doesn't update in driver panel
shutdown and restart driver, note temperature shows the more recent temperature

**Expected behavior**
Temperature reported to Indi and kstars should update periodically as temperature changes

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 LTS
 - Version kstars 3.6.7 indilib 2.0.4

**Log Files**

[log_17-47-48.txt](https://github.com/indilib/indi/files/12851764/log_17-47-48.txt)

closed 2023-10-10 by knro

gordoabc, 2023-10-10:
Updated Driver doesn't fix it.  Now getting serial write error"

2023-10-10T14:22:21: [ERROR] Serial write error: Timeout error 

[log_08-12-09.txt](https://github.com/indilib/indi/files/12858391/log_08-12-09.txt)

knro, 2023-10-10:
The change would not have caused this, this is another issue.

gordoabc, 2023-10-10:
I will hunt around and see what I can figure out

sonny486, 2023-10-10:
Do you have the latest firmware on your focuser? What version is it running?

gordoabc, 2023-10-11:
It is the latest firmware - the error is not reproducible, now I
The temperature is updating and I don’t get an error message. Maybe some random usb problem or something.  Will see if it is reliable.

gordoabc, 2023-11-04:
temperature updates are unpredictable - it can go a long time (with outdoor temperature changing) without an update.  The firmware is 3.5.13.  No repeat of the serial write error however.

gordoabc, 2023-11-09:
most recent change seems to fix the update frequency problem

commit 7f9f9645 (2023-10-10): Fix issue with temperature update. Fixes #1951