indilib/indi #1959: ZWO CCD driver alters biases and darks

FalCT60, 2023-11-02:
**Describe the bug**
ZWO CCD driver applies a WB correction to all darks and biases with ASI183MC non cooled.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Takes darks and biases with ASI183MC-nc and Kstars 3.6.6
2. Compute statistics on the series of files
3. Values for R, G and B ought to be the same, but in fact they get multiplied by a factor

**Expected behavior**
Darks and biases are not RGB pictures, but grey tones.
Attached is a spreadsheet showing the computing of a series of biases at gain 120 and offsets from 10 to 50 step 5, with SM, the other with NINA.
[offset.ods](https://github.com/indilib/indi/files/13242944/offset.ods)
Here is [the link](https://we.tl/t-yXlBG2GYsW) to the files used for the computation, so they can be analysed.

**Desktop (please complete the following information):**
 - OS: Stellarmate raspbian 64
 - Version: Kstars 3.6.6, SM OS 2.6.60

FalCT60, 2023-11-03:
Should this help : at gain 120 / offset 10, median values are 704 for green channel, but 731 for red and 1336 for blue.
All other values are affected too : sigma, MAD, etc.
this affects preprocessing and gives awful results.
This is not the camera's fault, but the driver's, as if I use another soft the darks and biases do not suffer those problems, all values are consistent.

MattBlack85, 2023-11-03:
I believe this is an old "issue", INDi sets the WB to these values:

- WB_R 52
- WB_B 95

You can set them both to 50 and calibration frames should come out good

knro, 2023-11-04:
INDI does not change these values, they're ready from camera on startup.

FalCT60, 2023-11-04:
@MattBlack85 yes, as stated [there](https://www.indilib.org/individuals/devices/cameras/zwo-optics-asi-cameras.html), those values must be set to 52 and 95. And this is what happens if I click on Default in options tab.
I was suggested to replace them by 50, and I did some computing that led to values of resp. 50.059 and 50.079 - so, 50.
The problem is - and it will be adressed in another bug report - that the values that I type in the Controls tab fields get sometimes overriden, despites being saved.
Yes, they remain while the system runs, but as soon as I restart my rPI they may have been restored to their original values.
It has appened more than five times today during my testings for WB_R, WB_B, Offset and Bandwidth, though I did have saved the config each time. As it did happen previously with Gain.
@knro do you mean they are embedded in the camera rom, and they are read on connection ?
Because, now that I have set them to 50 and saved the config, if I click to the Default button in Options tab, they will be restored and overrride my own settings. So, they must obviouly be read from somewhere.

knro, 2023-11-25:
No, there is a configuration file under ~/.ZWO as well that saves these settings in ZWO own format.

closed 2023-11-25 by knro