indilib/indi #1960: Explorestars Exos 2 PMC Eight refuses to connect to connect to EKOS over serial

matthearn, 2023-11-03:
I haven't used my telescope setup for about 18 months, I unpacked it recently and it was fine, updated kstars and EKOS, now my mount won't connect to EKOS.

See Logs:

[2023-11-03T14:22:08.770 GMT INFO ][                org.kde.kstars] - Welcome to KStars 3.5.7 Stable
[2023-11-03T14:22:08.770 GMT INFO ][                org.kde.kstars] - Build: 2022-05-04T07:33:10Z
[2023-11-03T14:22:08.770 GMT INFO ][                org.kde.kstars] - OS: "ubuntu"
[2023-11-03T14:22:08.770 GMT INFO ][                org.kde.kstars] - API: "x86_64-little_endian-lp64"
[2023-11-03T14:22:08.770 GMT INFO ][                org.kde.kstars] - Arch: "x86_64"
[2023-11-03T14:22:08.770 GMT INFO ][                org.kde.kstars] - Kernel Type: "linux"
[2023-11-03T14:22:08.770 GMT INFO ][                org.kde.kstars] - Kernel Version: "6.2.0-36-generic"
[2023-11-03T14:22:08.770 GMT INFO ][                org.kde.kstars] - Qt Version: 5.15.3
[2023-11-03T14:22:08.942 GMT DEBG ][                org.kde.kstars] - Opened the User DB. Ready.
[2023-11-03T14:22:09.138 GMT INFO ][                org.kde.kstars] - Processing  "unnamedstars.dat" , HTMesh Level 3
[2023-11-03T14:22:09.138 GMT INFO ][                org.kde.kstars] -   Sky Mesh Size:  512
[2023-11-03T14:22:09.155 GMT INFO ][                org.kde.kstars] - Loaded DSO catalog file:  "unnamedstars.dat"
[2023-11-03T14:22:09.342 GMT WARN ][                org.kde.kstars] - "Star HD20,794 not found."
[2023-11-03T14:22:09.351 GMT WARN ][                org.kde.kstars] - "Star HD98,230 not found."
[2023-11-03T14:22:09.360 GMT INFO ][                org.kde.kstars] - Loaded DSO catalogs.
[2023-11-03T14:22:09.435 GMT INFO ][                org.kde.kstars] - Loading comets
[2023-11-03T14:22:09.516 GMT DEBG ][                org.kde.kstars] - Paths to color scheme :  ("/usr/share/kstars/themes")
[2023-11-03T14:22:09.570 GMT DEBG ][                org.kde.kstars] - Reporting new timestep value:  60
[2023-11-03T14:22:09.721 GMT CRIT ][                     kf.xmlgui] - Shortcut for action  "get_data" "Download New Data..." set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
[2023-11-03T14:22:09.722 GMT CRIT ][                     kf.xmlgui] - Shortcut for action  "timestep_increase_units" "Increase Time Scale" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
[2023-11-03T14:22:09.722 GMT CRIT ][                     kf.xmlgui] - Shortcut for action  "timestep_decrease_units" "Decrease Time Scale" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
[2023-11-03T14:22:09.836 GMT DEBG ][                org.kde.kstars] - Daylight Saving Time inactive
[2023-11-03T14:22:09.836 GMT DEBG ][                org.kde.kstars] - Next Daylight Savings Time change (Local Time):  "Sun Mar 31 01:00:00 2024 GMT"
[2023-11-03T14:22:09.836 GMT DEBG ][                org.kde.kstars] - Next Daylight Savings Time change (UTC):  "Sun Mar 31 01:00:00 2024 GMT"
[2023-11-03T14:22:09.964 GMT DEBG ][                org.kde.kstars] - Daylight Saving Time inactive
[2023-11-03T14:22:09.964 GMT DEBG ][                org.kde.kstars] - Next Daylight Savings Time change (Local Time):  "Sun Mar 31 01:00:00 2024 GMT"
[2023-11-03T14:22:09.964 GMT DEBG ][                org.kde.kstars] - Next Daylight Savings Time change (UTC):  "Sun Mar 31 01:00:00 2024 GMT"
[2023-11-03T14:22:09.964 GMT DEBG ][                org.kde.kstars] - Starting the timer
[2023-11-03T14:22:10.055 GMT DEBG ][                org.kde.kstars] - Date/Time is: "Fri Nov 3 14:22:09 2023 GMT"
[2023-11-03T14:22:10.055 GMT DEBG ][                org.kde.kstars] - Location: "Dover, United Kingdom"
[2023-11-03T14:22:10.056 GMT DEBG ][                org.kde.kstars] - TZ0: 0 TZ: 0
[2023-11-03T14:22:10.056 GMT WARN ][                org.kde.kstars] - Current icon theme is "Yaru"
[2023-11-03T14:22:10.093 GMT DEBG ][                org.kde.kstars] - "Black Body"  ::  "/usr/share/kstars/themes/blackbody.colors"
[2023-11-03T14:22:10.240 GMT DEBG ][                       default] - glibc >= 2.1 detected.  Using GNU extension sincos()
[2023-11-03T14:22:10.335 GMT WARN ][                       default] - QImage::scaled: Image is a null image
[2023-11-03T14:22:10.838 GMT INFO ][                org.kde.kstars] - Loading comets
[2023-11-03T14:22:39.158 GMT DEBG ][           org.kde.kstars.ekos] - Resetting Ekos Manager...
[2023-11-03T14:22:39.175 GMT INFO ][           org.kde.kstars.ekos] - "Starting INDI services..."
[2023-11-03T14:22:39.175 GMT DEBG ][           org.kde.kstars.indi] - INDI: Starting local drivers...
[2023-11-03T14:22:39.176 GMT DEBG ][           org.kde.kstars.indi] - Starting INDI Server:  ("-v", "-p", "7624", "-m", "1024", "-r", "0", "-f", "/tmp/indififo8bc0d4f7") -f "/tmp/indififo8bc0d4f7"
[2023-11-03T14:22:39.182 GMT DEBG ][           org.kde.kstars.indi] - INDI Server Started?  true
[2023-11-03T14:22:39.182 GMT DEBG ][           org.kde.kstars.indi] - INDI: INDI Server started locally on port  7624
[2023-11-03T14:22:39.182 GMT DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_pmc8_telescope"
[2023-11-03T14:22:39.183 GMT DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_simulator_ccd"
[2023-11-03T14:22:39.183 GMT DEBG ][           org.kde.kstars.indi] - Adding managed driver "PMC8"
[2023-11-03T14:22:39.183 GMT DEBG ][           org.kde.kstars.indi] - Adding managed driver "CCD Simulator"
[2023-11-03T14:22:39.183 GMT DEBG ][           org.kde.kstars.indi] - INDIListener: Adding a new client manager to INDI listener..
[2023-11-03T14:22:39.183 GMT DEBG ][           org.kde.kstars.indi] - INDI: Connecting to local INDI server on port  7624  ...
[2023-11-03T14:22:39.287 GMT DEBG ][           org.kde.kstars.indi] - INDI: Connecting to local INDI server on port  7624  ...
[2023-11-03T14:22:39.287 GMT DEBG ][           org.kde.kstars.indi] - INDI server connected.
[2023-11-03T14:22:39.287 GMT DEBG ][           org.kde.kstars.indi] - Connection to INDI server is successful
[2023-11-03T14:22:39.288 GMT INFO ][           org.kde.kstars.ekos] - "INDI services started on port 7624."
[2023-11-03T14:22:39.289 GMT DEBG ][           org.kde.kstars.indi] - Received new device CCD Simulator
[2023-11-03T14:22:39.289 GMT DEBG ][           org.kde.kstars.indi] - Received new device ES EXOS2-GT PMC-Eight
[2023-11-03T14:22:39.293 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-11-03T14:22:39: startup: /usr/bin/indiserver -v -p 7624 -m 1024 -r 0 -f /tmp/indififo8bc0d4f7"
[2023-11-03T14:22:39.293 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-11-03T14:22:39: listening to port 7624 on fd 5"
[2023-11-03T14:22:39.293 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-11-03T14:22:39: Local server: listening on local domain at: @/tmp/indiserver"
[2023-11-03T14:22:39.293 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-11-03T14:22:39: FIFO: start indi_pmc8_telescope -n \"ES EXOS2-GT PMC-Eight\""
[2023-11-03T14:22:39.293 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-11-03T14:22:39: With name: ES EXOS2-GT PMC-Eight"
[2023-11-03T14:22:39.293 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-11-03T14:22:39: FIFO: Starting driver indi_pmc8_telescope"
[2023-11-03T14:22:39.294 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-11-03T14:22:39: Driver indi_pmc8_telescope: pid=7246 rfd=9 wfd=9 efd=10"
[2023-11-03T14:22:39.294 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-11-03T14:22:39: FIFO: start indi_simulator_ccd"
[2023-11-03T14:22:39.294 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-11-03T14:22:39: FIFO: Starting driver indi_simulator_ccd"
[2023-11-03T14:22:39.294 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-11-03T14:22:39: Driver indi_simulator_ccd: pid=7247 rfd=11 wfd=11 efd=12"
[2023-11-03T14:22:39.294 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-11-03T14:22:39: Driver indi_simulator_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD"
[2023-11-03T14:22:39.294 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-11-03T14:22:39: Driver indi_simulator_ccd: snooping on Telescope Simulator.EQUATORIAL_COORD"
[... 171 lines cut ...]
[2023-11-03T14:23:21.255 GMT DEBG ][           org.kde.kstars.indi] - < ES EXOS2-GT PMC-Eight >: < TELESCOPE_TRACK_MODE >
[... cut ...]
[2023-11-03T14:23:21.284 GMT DEBG ][           org.kde.kstars.indi] - < ES EXOS2-GT PMC-Eight >: < TELESCOPE_TIMED_GUIDE_NS >
[2023-11-03T14:23:21.286 GMT DEBG ][           org.kde.kstars.indi] - < ES EXOS2-GT PMC-Eight >: < TELESCOPE_TIMED_GUIDE_WE >
[2023-11-03T14:23:21.286 GMT INFO ][           org.kde.kstars.ekos] - "Guider port from ES EXOS2-GT PMC-Eight is ready."
[2023-11-03T14:23:21.288 GMT DEBG ][           org.kde.kstars.indi] - < ES EXOS2-GT PMC-Eight >: < Firmware >
[2023-11-03T14:23:21.290 GMT INFO ][           org.kde.kstars.indi] - ES EXOS2-GT PMC-Eight :  "[INFO] Debug is enabled. "

matthearn, 2023-11-05:
is there any further investigation I can do to help resolve this issue? I am desperate to get back to having a working astro rig.

knro, 2023-11-06:
Looks like it's using quite old KStars (and most likely old INDI as well). Though your log above actually indicates that it successfully connected to the mount.

closed 2024-03-03 by knro