indilib/indi #1980: Star Adventurer GTI not droven

FalCT60, 2023-12-20:
**Describe the bug**
I cannot drive my mount with new 1.8 SM OS. Commands are either inactive or greyed out.

**To Reproduce**
Exact steps to reproduce the behavior.
1. set profile including SA GTI (found as Sky adventurer in the list of mounts)
2. connect
3. the mount cannot be (un)parked nor driven

**Expected behavior**
Everything that ought to be possible with a mount plugged to the system

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: SM OS 1.8 on RPi5

**Log Files**
[2023-12-18.zip](https://github.com/indilib/indi/files/13730660/2023-12-18.zip)

knro, 2023-12-21:
`[2023-12-18T22:42:41.452 CET INFO ][           org.kde.kstars.indi] - SkyAdventurer GTi :  "[INFO] Mount is unparked. "`

Communication looks good in the logs and it also indicates that it is unparked. As far as the logs show, there are no issues.

FalCT60, 2023-12-21:
I am afraid there may be a huge difference between what looks and what is.
If you look at the Indi side, yes all seems good, as shown on Mount-Indi attached picture.
But, if you look at the Ekos side, this is another story, as shown on Mount-Ekos attachment.
And the same with the tab app : none of the commands are switchable (sorry, no screenshot there, but you must believe me).
[image]
[image]

FalCT60, 2023-12-21:
Well, half a week spent on this problem before I decided to fill a report, and I finally discovered the reason : for some unknown reason, the mount was not in the optical trains anymore.
[image]
I had to modify each of my settings to set it again.
Very strange, but at least it works now. 8-)

closed 2023-12-21 by FalCT60