indilib/indi #1982: indilib-driver-duino-2.0.5 libfirmata not being installed on linux

AlexPuchmayr, 2024-01-01:
**Describe the bug**

Dispite of compiling fine, indi-duino does not install libfirmata.so, yielding a broken binary /usr/bin/indi_duino.

indi-duino/CMakeLists.txt explicitly installs it only for APPLE (Why?). Reformulating the if statement will likely solve the problem.

**To Reproduce**
Compile drivers and install them from source

**Expected behavior**
libfirmata.so being installed properly

**Screenshots**
N/A

**Desktop (please complete the following information):**
gentoo linux, amd64

**Log Files**
ldd /usr/bin/indi_duino
        linux-vdso.so.1 (0x00007ffed5da2000)
        libindidriver.so.2 => /usr/lib64/libindidriver.so.2 (0x00007f43aee00000)
        **libfirmata.so => not found**
        libstdc++.so.6 => /usr/lib/gcc/x86_64-pc-linux-gnu/13/libstdc++.so.6 (0x00007f43aea00000)
        libgcc_s.so.1 => /usr/lib/gcc/x86_64-pc-linux-gnu/13/libgcc_s.so.1 (0x00007f43af086000)
        libc.so.6 => /usr/lib64/libc.so.6 (0x00007f43ae820000)
        libusb-1.0.so.0 => /usr/lib64/libusb-1.0.so.0 (0x00007f43af067000)
        libnova-0.16.so.0 => /usr/lib64/libnova-0.16.so.0 (0x00007f43ae200000)
        libcfitsio.so.10 => /usr/lib64/libcfitsio.so.10 (0x00007f43ade00000)
        libz.so.1 => /usr/lib64/libz.so.1 (0x00007f43af04b000)
        libjpeg.so.62 => /usr/lib64/libjpeg.so.62 (0x00007f43aed42000)
        libfftw3.so.3 => /usr/lib64/libfftw3.so.3 (0x00007f43adc7c000)
        libtheoraenc.so.1 => /usr/lib64/libtheoraenc.so.1 (0x00007f43aed13000)
        libtheoradec.so.1 => /usr/lib64/libtheoradec.so.1 (0x00007f43af036000)
        libogg.so.0 => /usr/lib64/libogg.so.0 (0x00007f43af02b000)
        libm.so.6 => /usr/lib64/libm.so.6 (0x00007f43ae740000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f43af0ff000)
        libudev.so.1 => /usr/lib64/libudev.so.1 (0x00007f43ae6ff000)
        libatomic.so.1 => /usr/lib/gcc/x86_64-pc-linux-gnu/13/libatomic.so.1 (0x00007f43af021000)
        libbz2.so.1 => /usr/lib64/libbz2.so.1 (0x00007f43ae6ec000)
        libcap.so.2 => /usr/lib64/libcap.so.2 (0x00007f43ae1f4000)

knro, 2024-01-01:
I think this might be a Gentoo issue? No problems here under Ubuntu and also I think it's statically linked as I don't see libfirmata.so under ldd

AlexPuchmayr, 2024-01-03:
After some research, I think it works  on Ubuntu by coincidence.

In the CMakeLists.txt, its not specified whether the library is to be created STATIC or SHARED (missing keyword), so cmake uses ${BUILD_SHARED_LIBS}. If this is set to "ON" (which is the case on gentoo since a couple of years now), shared libraries are created, otherwise (which includes  BUILD_SHARED_LIBS being unset) it uses static libs.

AFAIK Ubuntu does not define  BUILD_SHARED_LIBS by default, hence we have this situation.  I also suppose, whoever maintains the APPLE part, had the same problem and fixed it by adding this if(APPLE) statement. 

IMHO, the clean way to solve it is adding STATIC to the CmakeLists.txt, i.e.

add_library(firmata **STATIC** ${firmata_SRCS})

Then, for APPLE, I think the extra installation of the lib is also no longer required.

sterne-jaeger, 2024-01-04:
Solved, change is merged. Thanks for reporting it!

closed 2024-01-17 by knro