indilib/indi #1983: Cannot connect to flip-flat with virtual serial port unless 'Drop RTS' checks are bypassed as follows

jfwells, 2024-01-01:
I'm using INDI on an RPi wih  stepper/relay hats. I've impemented my flip-flat as a local python daemon that spawns a local serial pty.

This works fine, but only if the "Drop RTS" hardware ioctl  checks in drivers/auxiliary/flipflat.cpp are bypassed 

I have simply commented out the two 'return false;' statements on lines 154 and 161.

Is this section really needed for 'real' flip-flats? if so, I obviously can't then submit this as a PR. 

Is there any way this could be mdified that would work for both 'real' and 'viertual' serial ports so that I can go back to using upstream INDI?

knro, 2024-01-02:
It's obviously required for the actual physical port otherwise it wouldn't be there. This driver does not get any update so you can safely fork and modify to fit your use case.

closed 2024-01-02 by knro

jfwells, 2024-01-02:
> It's obviously required for the actual physical port otherwise it wouldn't be there. This driver does not get any update so you can safely fork and modify to fit your use case.

Thanks. Yes, the key part is that it doesn't have to return false on failure immediately, as this prevents the virtual serial port from succeeding.

...guess I  need to figure out how to maintain a single forked driver without recompiling all INDI