indilib/indi #1990: ieaf-focus reversing direction doesn't work

vrascal, 2024-01-11:
**Describe the bug**
After turning on the reverse function, the actual focus direction does not change.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Desktop (please complete the following information):**
 - OS: Debian Bookworm 12
 - Desktop：xfce4
 - Kstars: 3.6.8
 - indi_ieaf_focus:  compile from source

**Log Files**
2024-01-11T14:25:32: [INFO] iEAF Focus parameters updated, focuser ready for use. 
2024-01-11T14:25:30: [ERROR] Invalid Response: focuser Reverse direction value (+027898027552#) 
2024-01-11T14:25:25: [INFO] iEAFFocus is online.

knro, 2024-01-12:
@Joe13905179063 Can you please check the issue?

Joe13905179063, 2024-01-14:
Thanks!   I test it works OK!  Please update iOptron iEAF the latest firmware  and try again.

vrascal, 2024-01-14:
> Thanks! I test it works OK! Please update iOptron iEAF the latest firmware and try again.

Thanks Reply.
I have tried to find and update the firmware.
Because the device is at the remote site, it will take some time to process.

closed 2024-01-14 by vrascal