indilib/indi #1992: Pegasus PPBA/M missing auto dew aggressiveness setting

ijessen, 2024-02-03:
**Describe the bug**
The PPBA/PPBM driver doesn't handle the auto dew "aggressiveness" setting.  It looks like support was added for this parameter in a later serial protocol update (Jan 2021), [see here](https://pegasusastro.com//wp-content/uploads/2022/05/Pocket_Powerbox_Advance_Serial_Command_Table.pdf).

knro, 2024-02-04:
Thank you for the report. Do you think you can work on a PR for this?

ijessen, 2024-02-05:
Yep - am planning to

ijessen, 2024-02-08:
I started on this but my PPBM doesn't seem to respond to the aggressiveness commands like the documentation says it should.  I have a request in to Pegasus for info.

[image]

ijessen, 2024-02-22:
Pegasus support finally responded that they will update their documentation at some point, hopefully soon

closed 2024-03-03 by ijessen