indilib/indi #1993: "Fast Exposure" option undocumented and unclear

databu, 2024-02-05:
**Describe the bug**
Using GPhoto CCD, there's a "Fast Exposure" option, including a seemingly related param "Fast Count". It's not documented and not clear what these options do.

**To Reproduce**
Try to find out what "Fast Exposure" and "Fast Count" options do.

**Expected behavior**
Find some docs explaining them.

NB: I looked up [the code](https://github.com/indilib/indi/blob/5176249078640d86142a0d6b361c2a3278c3c79b/libs/indibase/indiccd.cpp#L2688), and if I understand correctly, "Fast Exposure" lets the camera start the next exposure without waiting for the upload of the previous one to complete. However I'm not too sure about it, and also not what the count does.

There also seems to be a code smell: `UploadS[UPLOAD_LOCAL].s == ISS_ON` in the condition in line 2721 can never be true because it's contained in an `if` block with `if (UploadS[UPLOAD_LOCAL].s != ISS_ON)`.

knro, 2024-02-15:
@naheedsa Please add this property to the documentation as a standard property.

knro, 2024-02-15:
Fast Exposure is used to enable camera to immediately begin capturing the next frame once the previous frame data is downloaded from the camera. With Fast Exposure disabled, the driver have to wait until the client initiates the next capture request. To minimize the downtime, Fast Exposure can be enabled to trigger for a specific number of frame in Fast Exposure Count property.

Once the initial capture is started, the driver would continue to capture all necessary frames without waiting for client until the count reaches zero. The Fast Exposure Count is decremented after each fast exposure is complete.

Upload Mode affects Fast Exposure behavior depending on the mode selected. For LOCAL mode, where images are saved locally to disk, Fast Exposure is most efficient. For CLIENT or BOTH modes, the driver have to transmit the data over network to the client. If the upload time exceeds the requested exposure time, then Fast Exposure cannot reliably work and the driver would abort the exposure.

databu, 2024-02-15:
Thanks for the explanation. 

> With Fast Exposure disabled, the driver have to wait until the client initiates the next capture request.

Shouldn't the client initiate the next exposure immediately anyway after downloading the previous one (assuming no delay is configured)?

knro, 2024-03-03:
yes, but this waits for download from server to client first and then after that is done, the client starts. It's not as fast as native Fast Exposure. We added documentation for this to docs.indilib.org

closed 2024-03-03 by knro

databu, 2024-03-04:
Awesome, thank you!