indilib/indi #2026: SCT focuser interface compliance

ijessen, 2024-03-13:
When I added support for the Celestron SCT focuser to the celestron-aux driver in the other repo, I opened a PR to align that focuser to the interface definition - namely that absolute focus values start at 0 at the intrafocal limit and increase positively toward the extrafocal limit.

The existing SCT focus driver in this repo doesn't align to that definition.  The value ranges from an arbitraty minimum derived from calibration, and increases in the intrafocal direction (i.e. wrong) to a maximum.

This isn't such a big deal when using the recommended 1-pass linear autofocus algorithm with 0 AF overscan, since that routine always ends by *increasing* toward the final value.  However - AF overscan is on by default, which is exactly the wrong move for SCTs.  

I'm happy to work up a PR to align this driver with the focus interface definition, as I did w/ the 3rd party driver.  But where in that case I was adding a new function, here I would be changing the behavior (and the values) of an existing driver, which probably warrants some discussion first.  The new behavior could be made an option.

knro, 2024-03-14:
How would this fair with different clients? The driver should be conform to INDI standards and not necessary to how Ekos operates its autofocus.

ijessen, 2024-03-15:
The change I'm proposing *is* to align the focus driver with INDI standards - it is not currently.  Presumably this would benefit other INDI clients.  I completely get your point that the INDI drivers shouldn't be designed for Ekos only.  I guess what I'm saying is that the existing driver implementation both isn't ideal for Ekos (at least not with AF overscan set) and also isn't compliant with INDI standards.

The more I've looked into this driver - the more confused I get.  Backlash compensation, for example, seems to mostly replicate the Ekos AF overscan behavior (i.e. overshoot-then-retract applied to all moves in one direction) rather than a true backlash compensation (i.e. constant added to any move when reversing direction from previous).  The overscan behavior is beneficial for SCTs when imaging - but I would prefer standard backlash compensation (without overscan) for visual use.

knro, 2024-03-15:
If it is to align with INDI standards, then sure go ahead! It would be great. Would this mean it needs to be changed in 3 places?
1. Celestron GPS driver
2. Celestron SCT focuser standalone driver
3. Celestron AUX driver

ijessen, 2024-03-17:
OK I'll get started on - it'll take a little while since I don't have much time these days.

1. Celestron GPS driver
2. Celestron SCT focuser standalone driver
3. ~~Celestron AUX driver~~

Change will need to apply to both the Celestron GPS and SCT focuser drivers, but not the AUX driver, since that is already compliant.

closed 2024-05-24 by github-actions[bot]