indilib/indi #2028: Bug in fli_pdf focuser driver not handling AF Overscan properly

johneevans, 2024-03-15:
**Describe the bug**
The fli_pdp is quite an old driver and was probably written before AF Overscan was added to Focus. AF Overscan relies on the focuser always returning the current position as it moves to its requested position. The fli_pdp driver returns FLIFocus.steps_remaing when a move is in flight (line 343) and FLIFocus.current_pos (line 353) when the move completes. This confuses Focus and renders AF Overscan unusable.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Perform AF Overscan move outward, e.g. from position 1000, set overscan=100 and move to 1050. The focuser should move first to 1150, when that completes, it should automatically move in to 1050.
3. The bug is that Focus will receive a position counting down to 0 and then trigger an inward move from 0 to -100 which will fail.

**Expected behavior**
As per 2. above.

**Screenshots**
See this thread for info... https://indilib.org/forum/ekos/14209-problem-with-af-overscan-to-handle-focuser-backlash.html?start=0#98273

**Desktop (please complete the following information):**
 - OS:
 - Version [e.g. 1.7.6, GIT..etc]

**Log Files**
See this thread for more info... https://indilib.org/forum/ekos/14209-problem-with-af-overscan-to-handle-focuser-backlash.html?start=0#98273

crfrancis, 2024-03-15:
Another observation with this driver is that when Autofocus fails and tries to move to its original position, the focuser actually moves to position 0, which is way out of the correct zone of about 50000, and from which it cannot recover. I don't know for sure, but is likely due to the same bug.

knro, 2024-03-17:
Thank you for the report. This is an issue in 3rd party repo, not INDI core. However, I believe I submitted a fix there. Please recompile and test again.

closed 2024-03-17 by knro

crfrancis, 2024-03-17:
Thanks Jasem,

I would love to build it but, as reported in this thread https://www.indilib.org/forum/development/14443-build-failed-with-indiserver-py-error.html
I have a problem completing the build. I am getting the following fail with indiserver.py

/Users/rfrancis/AstroRoot/craft-root/etc/blueprints/locations/craft-blueprints-kde/libs/_mac/indiserver/indiserver.py failed:
type object 'super' has no attribute ‘install'

This is a macOS build, using Rob Lancaster’s script. It’s cloning the git repo so even if I fix it locally (which I can’t as I don’t know Python) it will still get this version, as I understand it.

Thanks,
Richard

> 
> 
> Thank you for the report. This is an issue in 3rd party repo, not INDI core. However, I believe I submitted a fix there. Please recompile and test again.

-- 
------------------------------------------------------
Richard FRANCIS

GERSPATIAL SAS
Le Bourdieu
32480 La Romieu

***@***.***
+33 (0)6 32 54 65 07
www.gerspatial.com
www.gersastronomie.com
------------------------------------------------------