indilib/indi #2030: JMI MotoFocus Plus connection error

liquiduniverse, 2024-03-15:
**Describe the bug**
My JMI Motofocus Plus is showing in the OS StellarMate Dashboard hardware as USBFocusV3 with driver indi_usbfocusv3_focus. Indi shows the full default port for the USBFocusV3 is /dev/serial/by-id/usb-SparkFun_HCU-f00. I then changed the port to /dev/ttyACM0, per the guidance instructions. I receive the exact results for both ports.  I receive error: retrieving data from USBFocusV3, please insure powered on and on correct port. It retrys 6 times with no success and error: No Valid Resync Response. Thank you in advance for your help.

I expect Indi to connect to the JMI Motofocus Plus/USBFocusV3

OS: Stellarmate OS on Intel mini pc

Logfile is added as zip file.
[log_17-34-48.zip](https://github.com/indilib/indi/files/14621193/log_17-34-48.zip)

jpaana, 2024-03-16:
I doubt it very much that JMI focuser would be compatible with USB_FocusV3 and I don't know where Stellarmate gets that from. Maybe they use similar microcontrollers and USB ids, but USB_FocusV3 has their own vendor id and not SparkFun which is a popular Arduino-style DYI board.

liquiduniverse, 2024-03-16:
Is there anything that can be done to get it working?

liquiduniverse, 2024-03-16:
Here is a snippet from the Stellarmate Dashboard for the JMI Motofocus Plus.
[image]

liquiduniverse, 2024-03-16:
Is there anything that can be done with how Stellarmate is recognizing the JMI Motofocus Plus as USBFocusV3, or are you able to properly identify the JMI Motofocus Plus and create a proper driver to have it correctly recognized by Indi.  Thank you.

jpaana, 2024-03-16:
All those ids are different than what USB_FocusV3 uses so it looks like an error in the StellarMate info, not just accidental ID overlap. I don't know if there's a driver for JMI Motorfocus Plus, but creating one would be fairly straight forward if the protocol is documented. Obviously having a device to test with would help too. However there seems to be a JMI SmartFocus driver which might be worth trying as well. If you have Windows drivers for the device, it would also be possible snoop the traffic for protocol info like I did for USB_FocusV3 originally.

liquiduniverse, 2024-03-16:
I’ve searched and am unable to locate a JMI Motofocus Plus driver.  I tried using the SmartFocus driver and that did not work.  I’m far from being able to successfully snoop the traffic on a Windows driver (I have no idea what that even means).  The only thing I have is the JMI Motofocus Plus software that I used with ASCOM in Windows.

What is involved with moving forward to having you create a JMI Motofocus Plus driver and without you having the device?

jpaana, 2024-03-16:
I took a closer look at what kind of device you actually have and it seems to be a so called relative focuser and not absolute focuser so that rules out the SmartFocus driver even they are from the same manufacturer. So it would require writing a new driver, but that is fairly simple based on one of the similar drivers we already have like nFocus, but requires info about the serial protocol so the commands and responses would match.

I couldn't find any documentation on that so it would require capturing the serial port traffic starting from connecting with the Windows program and using all the necessary features and document all the steps so they can be correlated with the serial traffic. Like: connect - move focuser inwards - move focuser outwards - change some setting - and so on. I did that with the USB_FocusV3 and it was quite straight forward to reverse engineer the protocol that way and your focuser has even less functionality and hopefully complexity. For Windows there are some free programs that can capture serial port traffic quite easily. Easiest is probably PortMon (https://learn.microsoft.com/en-us/sysinternals/downloads/portmon).

If you can provide such a capture, I can take a look and probably make the driver fairly easily, but can't test it without the device obviously. If you have compilation environment yourself, you could then compile the driver yourself, but I can also provide x86_64 binary or maybe Jasem as he's the main StellarMate guy. That said, I have no affiliation with StellarMate, I'm just a volunteer like most INDI developers so no promises or commitments, but it seems easy enough that I should have time to do it fairly quickly as I have done quite a few drivers already. And thanks to open source I can do mostly a copy & paste with some search & replace from an existing driver :)

liquiduniverse, 2024-03-17:
First of all, I want to thank you for all you have done so far by looking into my issue with the JMI MotoFocus Plus and the Indi driver issue, and for providing the instructions to move forward. I will work on this project per your guidance and will keep you posted on my progress.  Be well and enjoy the rest of your weekend.

liquiduniverse, 2024-03-17:
Hi Jarno.  I'm considering mailing you the JMI MotoFocus Plus for you to develop the driver. Is this possible if I pay all the shipping fees?

jpaana, 2024-03-18:
That's probably not necessary, and would be quite a hassle with customs etc. unless you live in EU. I can go through the steps required for capturing the serial traffic from Windows or we can setup a remote session with TeamViewer or similar, assuming we can find a common time :) I live in UTC+2 (soon UTC+3 due to DST).

I haven't used PortMon for a while and had some issues getting it to run with my Windows 11 machine, but got https://freeserialanalyzer.com/ to work. It can dump the traffic to window and also to some kind of replay log file.

jpaana, 2024-03-18:
It shouldn't make a difference if the controlling software is ASCOM driver or a standalone program, serial communication with the device should be the same. I think (not sure) that ASCOM expects focuser to be an absolute one, ie. being controlled with absolute step counts and being able to recall the exact position. A relative focuser is kind of control for the buttons so it just moves in and out and is controlled by time (and possibly speed) so it's not as accurate to control as an absolute one. Autofocusing is easier with absolute focuser, but I have a vague recollection that Ekos also has some support for relative focusers with the one-pass linear algorithm.

Seems PortMon requires a 32-bit Windows version which explains why it didn't work with my Windows 11 anymore. I had used it back in the day with a Windows XP machine and there it worked fine. But the Device Monitoring Studio I linked above worked fine. It's quite limited in free functionality and nags about paid features, but should be enough for this case as we don't need any fancy protocol decoders or such.

liquiduniverse, 2024-03-27:
Update…originally, when I decided to get a focuser, it was noted that the Celestron focuser was not compatible with the Celestron GPS or models before 2006.  Since I own on older model Celestron GPS, I did not purchase the Celestron focuser and ended up purchasing the JMI Motofocus Plus.  Before I made this purchase, I discovered the early Celestron GPS model telescopes control board caused the telescope to be off 10 degrees during the alignment process (or something like that) and firmware updates weren’t capable, and to resolve this problem you could install an updated control board, which I did.  Now, while going through this focuser problem with the JMI Motofocus Plus, I decided to go ahead and purchase the Celestron focuser, I did this because worse case would be it wouldn’t focus using the hand controller, but could use the computer to focus, which I was willing to live with.  Well, after installing the Celestron focuser, I have no incompatibility issues at all.  It installed on the telescope perfectly, it works through the hand controller perfectly and of course with Indi.  I want to thank you for communicating with me and assisting with the JMI Motofocus Plus, but now there is no need anymore.  Thanks for everything.

closed 2024-06-05 by github-actions[bot]